ArduPilot/ardupilot #18330: MSP_OSD_NCELLS and OSD_CELL_COUNT  should determine the number of cells only at the time of connecting the battery.

DarkSoulRL, 2021-08-14:
The number of battery cells is recalculated during the flight. This leads to an error in displaying the voltage of one cell on the OSD when the voltage drops under load, or when using Li Ion batteries, which can drain up to 2.5V  per cell.

**Version**

ArduPlane V4.1.0beta4 (26689aab)
Arducopter  V4.1.0beta5
**Platform**
[X  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
Matek H743 slim
Matek H743 wing

DarkSoulRL, 2021-08-28:
Yes, I often see on the DJI OSD that when the voltage on the 6S battery drops below 21V, the voltage of the cell starts to show 4.1V instead of 3.4V
Today we're going to fly and I'll try to record video from the DJI goggles screen.

DarkSoulRL, 2021-08-28:
The number of cells is calculated directly during the flight. Look at the cell voltage in the upper right corner. After a drawdown below 21B, ardupilot considers that there are 5 cells, not 6.

A 6S low-current battery, which is not fully charged, is specially connected so that the result can be seen immediately.

https://youtu.be/s9VP21Wt37A

IamPete1, 2021-08-28:
Hum, that is certainly wrong. Unfortunately I cant test the DJI OSD so easily. I wonder if DJI is using the cell count were passing. @yaapu maybe you have some insight?

yaapu, 2021-08-28:
@IamPete1 @DarkSoulRL I'm sure I tested this on DJI hardware at some point but I will test this again with a variable power supply, not sure I can go up to 6S though but should work just fine.

btw we should discard the MSP cell count parameters in favor of the OSD ones, at the time I wrote the MSP library OSD did not report cell count so it's time to refactor :-)

DarkSoulRL, 2021-08-28:
The same goes for the 4S battery. I use "auto" because I fly with both 4S and 6S batteries, depending on the requirements. At the same time, I change the propeller 7 inches or 6.

mike-tk, 2025-10-28:
Is this bug still actual? Because from the code it looks like cell count can go up in flight
https://github.com/ArduPilot/ardupilot/blob/fd95c32d886503709ce1a7bece72df2d7c66b17a/libraries/AP_OSD/AP_OSD_Screen.cpp#L1275