ArduPilot/ardupilot #18356: AC_Avoidance/AP_Proximity: Get wrong object when rotate  Copter (Radar 3 sectors)

framosnt, 2021-08-17:
## Bug report
I have the following problem using 3-sector radar (7,0,1), when the drone makes short rotations, up to 90 degrees, we have no problems, the obstacles represent correctly, but if the drone rotates more than 90 degrees , it gets lost, and registers obstacles in the same fixed location, generating a complete circle around the drone, making the drone stand still and have nowhere to go!

_Please describe the problem_

**Version**
_4.0.7 copter

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
quad

**Hardware type**
CUAV v5 - FMU2

https://youtu.be/evlGuVtDC24

rmackay9, 2021-08-17:
@framosnt, any chance you have a log file.  Also which lidar is being used?

rishabsingh3003, 2021-08-17:
@framosnt can you try 4.1 - beta?

framosnt, 2021-08-20:
> @framosnt, any chance you have a log file. Also which lidar is being used?

I used radar mr77 like Tera Ranger Tower 360 lidar (prx type 6). It can see 0, 1 and 7 sectors. Serial mode.

framosnt, 2021-08-20:
> @framosnt, any chance you have a log file. Also which lidar is being used?

Will find this log (Tlog is right ? )

framosnt, 2021-08-20:
> @framosnt can you try 4.1 - beta?

No, use 4.0.7 official and 4.0.9 !