ArduPilot/ardupilot #18359: from_rotation_matrix()

apm0x400, 2021-08-17:
Dear friend:
I thinks from_rotation_matrix() comment is misunderstanding。
 ardupilot/libraries/AP_Math/quaternion.h   Line70

` // return the rotation matrix equivalent for this quaternion
    void		from_rotation_matrix(const Matrix3<T> &m);`
this should return quaternion equivalent for this rotation matrix

peterbarker, 2021-08-17:
@apm0x400 It could be better.  Check the linked PR, please.

apm0x400, 2021-08-17:
OK,Thanks,also need to change libraries/AP_Math/quaternion.h

peterbarker, 2021-08-18:
> OK,Thanks,also need to change libraries/AP_Math/quaternion.h

Fair point, done.

There are many other descriptions in that file that could do with some help, however.

Feel free to raise PRs to fix this sort of thing rather than just raising issues - its not that much harder :-)

closed 2021-08-19 by peterbarker