ArduPilot/ardupilot #18376: Copter-4.1: occasional 68ms delays on Matek H7

rmackay9, 2021-08-18:
We have multiple cases of 68ms delays on the Matek H7 boards.

Report on 4.1 issues list. https://discuss.ardupilot.org/t/copter-4-1-0-beta7-has-been-released-for-beta-testing/74794/4

rmackay9, 2021-08-18:
Some things of note re this board:

- this board has a 16 bit timer
- INS_HNTCH_ENABLE = 1 (harmonic notch enabled)
- MOT_PWM_TYPE = 6 (PWM_TYPE_DSHOT600)
- OSD_TYPE = 1 (enabled)
- RNGFND1_TYPE = 20 (Benewake TFMini)
- RSSI_TYPE = 3 (Receiver)

chobitsfan, 2021-08-18:
I think this may be the same issue as  #18196 ?

rmackay9, 2021-08-18:
Yes, I agree, it's a duplicate issue so I'll close this one, thanks!

closed 2021-08-18 by rmackay9

rmackay9, 2021-08-19:
Here's another issue on the same topic: https://github.com/ArduPilot/ardupilot/issues/18383