ArduPilot/ardupilot #18379: SERVOn_MIN and SERVOn_MAX have no effect when using RCINn

drtrigon, 2021-08-18:
## Bug report

**Issue details**

`SERVOn_MIN` and `SERVOn_MAX` seem to have no effect when using `RCINn` in `SERVOn_FUNCTION`.

Example: I configure a gimbal e.g. SERVO8_FUNCTION=mount_tilt and SERVO9_FUNCTION=mount_roll and then use the min and max to configure the travel. When I now switch to SERVO8_FUNCTION=RCIN9 and SERVO9_FUNCTION=RCIN10 the given min and max do not apply anymore. Instead it just uses what comes from the RC receiver.

I think that is wrong behavior. Consider the fact that the RC channels do get calibrated so the calibrated range should be mapped to the configured output range. Currently it just seems to be kind of pass-trough behavior.

In my case I would like to use this to extend the PWM values to a range that my RC cannot generate (but the FC can).

**Version**
4.0.7

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad X

**Hardware type**
Matek H743 Wing

This might be related or the same issue than https://github.com/ArduPilot/ardupilot/issues/10488, https://github.com/ArduPilot/ardupilot/issues/9830

IamPete1, 2021-08-18:
It is by design to be direct passthrough. We could add a new set of output `RC x Scaled` where they could apply the min, max, trim and reversing.

It sounds like in your case your just trying to control the gimbal? How about using the `MNT_RC_IN_ROLL`/`MNT_RC_IN_TILT`/`MNT__RC_IN_PAN` params?

drtrigon, 2021-08-18:
>It is by design to be direct passthrough. We could add a new set of
>output `RC x Scaled` where they could apply the min, max, trim and
>reversing.

Yes, please!

Even though it is somehow strange to have to select the correct input in order to have the output to scale. But meanwhile I read through the other issues and do somehow understand the historic implications. So I like this proposal!

It would be great when the docu could contain some hints about these details, e.g. the pass-through behavior.

>It sounds like in your case your just trying to control the gimbal? How
>about using the `MNT_RC_IN_ROLL`/`MNT_RC_IN_TILT`/`MNT__RC_IN_PAN`
>params?

Yes, but I have some needs that ArduPilot cannot fullfil as I have to conclude from docu:
- I have a camera trigger that allows to preview, take pics AND take movies
- I would like to choose in mission mode whether gimbal and cam are manually controled or auto controled

If you have some hints on how to do this it would be very welcome!

drtrigon, 2022-03-18:
@IamPete1: What about bringing this to a dev call?

drtrigon, 2022-03-25:
@IamPete1: Can you explain to me what the label "good first issue" means? Do I have to take further action? Or what does happen next?

IamPete1, 2022-03-25:
@drtrigon It marks it as something relatively straight forward that a new developer could take on.

IamPete1, 2022-09-03:
Closed by https://github.com/ArduPilot/ardupilot/pull/21347 we now have output function RCIN1Scaled ... RCIN16Scaled

closed 2022-09-03 by IamPete1

drtrigon, 2022-09-05:
Brilliant!

Thanks @IamPete1 and everybody involved into #21347 !