ArduPilot/ardupilot #18408: Copter: mixed use of ap.using_interlock and motors->get_interlock()

rmackay9, 2021-08-21:
Copter has two ways of checking the motor interlock state. Sometimes it checks like this

> motors->get_interlock() && copter.ap.using_interlock

.. while other times it just checks like this

> motors->get_interlock()

Functionally I think we shouldn't need to check the ap.using_interlock and the mix means that we will get different parts of the system reacting differently if the user changes RCx_OPTION value (on multicopters) enabling/disabling interlock.

We've never had a user report of problems though.

peterbarker, 2021-08-23:
On Fri, 20 Aug 2021, Randy Mackay wrote:

> Copter has two ways of checking the motor interlock state. Sometimes it checks like this
>
>       motors->get_interlock() && copter.ap.using_interlock
>
> .. while other times it just checks like [...]

+1 for getting rid of using_interlock.  Notionally someone could be using 
a button or script to change that state - it doesn't need to be assigned 
to an RC switch.

rmackay9, 2021-08-23:
For anybody else reading this, remember that it is easy to make the change but the hard part is the testing.  The testing will take 10x as long as the development because whomever does it needs to go through and check each case through inspection and also with some actual testing in SITL.