ArduPilot/ardupilot #18430: default battery params not loading from ROMFS

tridge, 2021-08-24:
PR #17820 broke the loading of default values for analog pins and multipliers from defaults.parm embedded in ROMFS. When booting a board that sets these pins in defaults.parm we see this on the console:
```
Ignored unknown param BATT_VOLT_PIN from embedded region (offset=234)
Ignored unknown param BATT_VOLT_PIN from embedded region (offset=234)
Ignored unknown param BATT_CURR_PIN from embedded region (offset=251)
Ignored unknown param BATT_CURR_PIN from embedded region (offset=251)
Ignored unknown param BATT_VOLT_MULT from embedded region (offset=273)
Ignored unknown param BATT_VOLT_MULT from embedded region (offset=273)
Ignored unknown param BATT_AMP_PERVLT from embedded region (offset=296)
Ignored unknown param BATT_AMP_PERVLT from embedded region (offset=296)
```
ping @hendjoshsr71

tridge, 2021-09-15:
turns out it is not an issue. They do load when param sub-table is created

closed 2021-09-15 by tridge

hendjoshsr71, 2021-09-16:
Oh sorry I misunderstood what you meant here!

I thought the bug was about those warnings being generated in cases where
the param tree won't exist. And not wanting to bug the user for those.

 Not that the parameters weren't getting loaded when the analog monitor
does exist.

PR #18673 (2021-09-16): AP_Param: disable debug code by default
found on EU dev call
fixes #18430 
saves 756 bytes on F405