ArduPilot/ardupilot #18437: 4.1+ GPS GPA delta spikes

Sekilsgs2, 2021-08-24:
4.1+ versions have very unstable GPS task - if i'm using 234000 baudrate- i'm have many messages about gps health - fc cant reading data with 5hz.
If i'm using 57600 baud rate - more better.
I'm not using DMA for gps because this cause more warnings about gps health.
This 100% task scheduler problems - i'm think some tasks break  other tasks and we have delays.. this cause many problems which cause internal errors..
[image]

WickedShell, 2021-08-24:
We need to know more about the setup, what flight controller, what GPS unit, what port is it on, what other devices are on the other serial ports...

rmackay9, 2021-08-25:
Thanks very much for the report.  Normally we try to keep the support items in the support forums until they're confirmed bugs so could you raise an issue in the 4.1 category for Copter, Rover or Plane (whichever one you're using)?  Sorry to be strict but if we're not then the to-do list (which is what this issues list is) just turns into another support forum.  We can re-open this issue once we've confirmed it is a real bug and not a configuration or other setup issue.

Copter: https://discuss.ardupilot.org/c/arducopter/copter-41/159
Rover: https://discuss.ardupilot.org/c/ardurover/rover-41/158
Plane: https://discuss.ardupilot.org/c/arduplane/plane-4-1/160

Certainly for Copter and Rover I'm actively monitoring and tracking all reports.

I'm going to close this issue but I'll be monitoring for replies.

closed 2021-08-25 by rmackay9

Sekilsgs2, 2021-08-25:
Sorry but what the hell?

This is discuss - 
https://discuss.ardupilot.org/t/gps-instability-reinitializiation-during-flight-bn880-mambaf405/74184/29
this problem present in bugs list.

I'm, write about this problem many messages - and no one help me fix this - i'm try many variaints and different gps modules.
Current my setup - 
mamba f405 mk2
gps bn220
dalrc 4in1
compas lis3mdl
crossfire rx connected to USART1 (i'm enable DMA in hwdef for this port)

As you see from discuss - for me increasing buffers partial fix this issue, but again using DMA - more buggy..

this is my changes to hwdef - 
```
 # RC input defaults to timer capture
 PA10 TIM1_CH3 TIM1 RCININT PULLDOWN
-PA9 USART1_TX USART1 NODMA
+PA9 USART1_TX USART1

 # Alt config to allow RCIN on UART
-PA10 USART1_RX USART1 NODMA ALT(1)
+PA10 USART1_RX USART1 ALT(1)

 # SBUS inversion control pin, active low
 PC0 USART1_RXINV OUTPUT HIGH GPIO(78) POL(0)
@@ -49,12 +49,12 @@ PB10 USART3_TX USART3
 PB11 USART3_RX USART3

 # USART6
-PC6 USART6_TX USART6
-PC7 USART6_RX USART6
+PC6 USART6_TX USART6 NODMA
+PC7 USART6_RX USART6 NODMA
```

I'm using USART6 for gps without DMA.

In 4.0.7 i'm cant have this problem.

Perhaps the problem started from the moment when the corrections were made in the work of the usart driver in chibis core - i'm dont know - i'm not have time to learn all code..

And as you see new AP have many internal errors - for me this is main and the most important problem - 
and I still don't see that there is any progress associated with fixing these problems.
For example  - make motor/compas noise calibrating - and after finish i'm have internal error - ???

Maybe this not only GPS problem - but a common threading problem..