ArduPilot/ardupilot #18455: AttributeError: 'CubeBlack' object has no attribute 'with_can'

user456813, 2021-08-25:
I'm facing this problem after running Tools/environment_install/APM_install.sh.

```
Traceback (most recent call last):
  File "/path/ardupilot/modules/waf/waflib/Scripting.py", line 158, in waf_entry_point
    run_commands()
  File "/path/ardupilot/modules/waf/waflib/Scripting.py", line 251, in run_commands
    ctx = run_command(cmd_name)
  File "/path/ardupilot/modules/waf/waflib/Scripting.py", line 235, in run_command
    ctx.execute()
  File "Tools/ardupilotwaf/ardupilotwaf.py", line 170, in execute
    run_command(self.cmd)
  File "/path/ardupilot/modules/waf/waflib/Scripting.py", line 235, in run_command
    ctx.execute()
  File "Tools/ardupilotwaf/ardupilotwaf.py", line 117, in execute
    return execute_method(self)
  File "/path/ardupilot/modules/waf/waflib/Scripting.py", line 573, in execute
    return execute_method(self)
  File "/path/ardupilot/modules/waf/waflib/Build.py", line 231, in execute
    self.execute_build()
  File "/path/ardupilot/modules/waf/waflib/Build.py", line 244, in execute_build
    self.recurse([self.run_dir])
  File "/path/ardupilot/modules/waf/waflib/Context.py", line 286, in recurse
    user_function(self)
  File "/path/ardupilot/wscript", line 539, in build
    _build_recursion(bld)
  File "/path/ardupilot/wscript", line 489, in _build_recursion
    bld.recurse(d)
  File "/path/ardupilot/modules/waf/waflib/Context.py", line 286, in recurse
    user_function(self)
  File "/path/ardupilot/ardupilot/wscript", line 675, in build
    if bld.get_board().with_can:
AttributeError: 'CubeBlack' object has no attribute 'with_can'
```

I'm on ArduPlane-stable tag

krishna0303, 2021-08-25:
I would like to work on this issue @user456813

user456813, 2021-08-28:
@IamPete1, thank you for your replay

> I would guess you need to update sub-modules with:
git submodule update --init --recursive

I've deleted sub-modules entries from `.git/config` file, and `.git/modules`, and  `modules` directories.

then I run
```
git submodule update --init --recursive
Tools/environment_install/install-prereqs-ubuntu.sh
```

but, nothing changed at all.

IamPete1, 2021-08-28:
hum, not sure. Jump over to the discord https://discord.gg/x6abtzm3 post the result of a `./waf configure --board CubeBlack` maybe it cant find some of the required stuff.