ArduPilot/ardupilot #18589: Rover: cannot use steering mode without GPS

HefnySco, 2021-09-06:
If you switch to steering mode without GPS the rover simply does not mode at all.
I tried to trace back the reason and I found this [commit](https://github.com/ArduPilot/ardupilot/commit/53e7436525bc6250c136f7559712c6058b264e41#r55981002
)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[X] Rover
[  ] Submarine

rmackay9, 2021-09-06:
Rover's steering mode requires a velocity estimate so I would not expect it to work without a GPS (or some other source of position estimate).  I'm surprised that you were able to enter steering mode without a GPS and if the vehicle was already in steering mode then an EKF failsafe should have triggered when it lost GPS.

HefnySco, 2021-09-06:
I was able to go in steering mode until this particular line has been added. 
So if I want to move my vehicle and enable Obstacle Avoidance indoor with no GPS. What mode should I use ?

rmackay9, 2021-09-06:
@HefnySco,

I think one of these Non-GPS systems is required: https://ardupilot.org/rover/docs/common-non-gps-navigation-landing-page.html.  So for an indoor rover the easiest might be wheel encoders.

Object avoidance with no position or velocity estimate didn't work before this change either I think right?

HefnySco, 2021-09-06:
https://youtu.be/nAjDFAATC5U This was at home & without GPS

HefnySco, 2021-09-06:
Actually I discovered this issue by chance yesterday when I  updated my local rover version and it suddenly stopped.
I was trying to work on [this PR](https://github.com/ArduPilot/ardupilot/pull/18100).

HefnySco, 2021-09-06:
hence the sensor is calculating distance from object then we can use the object as a beacon and calculate speed based on changing in distance from object. I just want to avoid hitting the wall or something similar.
@rmackay9 what do you think ?

rmackay9, 2021-09-06:
@HefnySco,

Steering mode can't really work without a velocity estimate because the pilot is controlling lateral acceleration and this can only be calculated using the vehicle's speed.

Simple avoidance also requires the vehicle's speed because we need to be able to calculate whether we have enough time to slowdown or not before hitting the obstacle.

I don't think that calculating the vehicle's speed from a change in lidar distance is going to be very reliable.  I can imagine that it can work in some very clean environments but I don't think it will be robust enough to be used by most users.

HefnySco, 2021-09-06:
@rmackay9 Thank you for clarification.

closed 2021-09-06 by HefnySco