ArduPilot/ardupilot #18593: Plane: `LAND_THEN_NEUTRL` only works on aileron, elevator and rudder

IamPete1, 2021-09-06:
It should work on all control surfaces, elevons, VTAL, D-Spoiler, FLAPS ect. 

https://github.com/ArduPilot/ardupilot/blob/0ee8b452ab76306d83c6252bcb9dc458bd7380bb/ArduPlane/servos.cpp#L872-L888

vrquaeler, 2021-09-07:
Hi Pete, 
It works on elevons. I use this for years. From my understanding "k_aileron" and "k_elevator" are inputs for the "channel_function_mixer (). 
Please feel free to correct me if I am wrong.

IamPete1, 2021-09-07:
@vrquaeler elevons should work for the neutral position option but will not work for the zero PWM option. I guess that is what you have been using, `LAND_THEN_NEUTRL` = 1?

The other control surfaces, such as flaps, it will just not work at all.

vrquaeler, 2021-09-07:
@IamPete1 

You are right, I use the option:

> LAND_THEN_NEUTRL = 1

magicrub, 2021-09-08:
Yup, I totally agree. This could use some improvement since it hasn't been touched after I originally wrote it half a decade ago :)

Another concern is... What do these options do to UAVCAN servos?

shiv-tyagi, 2021-09-16:
Hey @IamPete1 ! This is the first time I am contributing to the ardupilot codebase (apart from documentation). I would love to work on this issue. Being a newbie, I don't exactly know what this issue is talking about. But I am getting a rough idea that we have to do same thing which we are currently doing (in the existing code) with aileron, elevator, rudder to elevons, vtail, flaps etc. too. Am I going in right direction?

IamPete1, 2021-09-16:
@shiv-tyagi exactly. Probably the nicest way to do this would be to add a "is_control_surface" method to `SRV_Chanels`. Much like the should estop method we currently have.

https://github.com/ArduPilot/ardupilot/blob/17f8b0b11a7ce3105a9c0cdff233b0a321f3992d/libraries/SRV_Channel/SRV_Channel.cpp#L216

The in plane servos.ccp you can go over all the output channels and apply to any that are control surfaces. 

Jump on the discord for help, https://discord.gg/e2nhvZhU

shiv-tyagi, 2021-09-16:
@IamPete1 I tried to do this as suggested by you. Please review if my changes serve the purpose.

closed 2021-09-18 by magicrub