ArduPilot/ardupilot #18602: AP_RangeFinder: handle lightware SF11 output of 130m on loss of signal

rmackay9, 2021-09-07:
The lightware SF11C (at least) outputs 130m "on lost signal" so we should handle this in the AP_LightwareSerial driver.

Here's the relevant line in [the SF11 datasheet](https://www.documents.lightware.co.za/SF11%20-%20Laser%20Altimeter%20Manual%20-%20Rev%2010.pdf):

[image]

As mentioned by @wynn4 this issue https://github.com/ArduPilot/ardupilot/issues/7756 if we don't handle this out-of-range value then it can incorrectly be summed with good values and lead to an incorrect result.

Unfortunately this means we need to go through all the lightware datasheets for other products that use the same serial protocol to see if they are consistent (I'll bet they're not!).

Shyam-Sunder-saini, 2021-09-27:
i want work on it

shiv-tyagi, 2021-10-04:
@Shyam-Sunder-saini Are you still working on this issue?

Shyam-Sunder-saini, 2021-10-06:
> Yes, @Shyam-Sunder-saini <https://github.com/Shyam-Sunder-saini> I'm also
> wondering if you are still planning to work this issue.

Yes why not i am super excited for this

shiv-tyagi, 2021-10-16:
@rmackay9 I would like to work on this issue. I have gone through documentation of almost all Lightware lidars and unfortunately I did not find any mention of "on lost signal" in any of the documentations except the one for SF11. I doubt if this "on lost signal" thing is specific for SF11 (being a newbie, I am not much sure about it). I propose to add parameter in Lightware library that helps us to differentiate between different models and handle 130 m output for SF11. Adding parameter will also help us to implement other model specific features (if any) in future. Your views about it?

shiv-tyagi, 2021-10-19:
After a lot of searching, I found out following things:-
- "on signal lost" outputs different values for different models.
- On some models this setting can be changed, on some it can't.

We can either add a parameter to differentiate between different models and handle settings for that lidar or we can ask lidars about their models using commands and then handle the behaviour for each model.

Currently, I don't have any hardware to test my results. This would make it difficult for me to fix this. If someone want to work on this issue, here is the reply of a mail I wrote to lightware customer support which might help.
[image]
This tells about other lidar models which have this setting (apart from SF11).

wynn4, 2021-11-05:
@shiv-tyagi Thank you for providing this info from LightWare. I have some hardware (although only one of these rangefinders--the SF11/C) and so I'm going to take a whack at addressing this issue.

My thinking is that we avoid extra talking back/forth with the rangefinder, and also avoid adding an extra parameter. Instead, we do some extra checks using these known "lost signal" flags. I've implemented this and done a successful build of my solution. I'll test soon using the hardware I have, and if successful, I'll submit a PR.

wynn4, 2021-11-09:
PR #19191

hendjoshsr71, 2021-12-26:
This was fixed with PR #19191.

Thank you @wynn4 !!

closed 2021-12-26 by hendjoshsr71