ArduPilot/ardupilot #18608: Possible to simulate "standard plane" in gazebo with sim_vehicle?

RotorBoy2020, 2021-09-08:
Is it possible to simulate a standard plane in gazebo with ardupilot? I am trying to simulate the [some of the planes in this video](https://www.youtube.com/watch?v=orMXVby-tSI)

I have been following this [tutorial](https://github.com/Intelligent-Quads/iq_tutorials/blob/master/docs/installing_gazebo_arduplugin.md)
Using this ardupilot [gazebo plugin ](https://github.com/SwiftGust/ardupilot_gazebo)

I am able to run arduplane sitl via sim_vehicle.py, connect the sitl vehicle to QGC, and connect it with MAVROS, but the gazebo plane models I have do not move (I grabbed them from the PX4 gazebo models folder). 

My questions is the following: Where can I find the gazebo models for simple cub like planes and how can I integrate them into gazebo? What needs to change about these files so that sim_vehicle.py can communicate with the gazebo model? 

If I run the same commands with a iris quadcopter, I am able to get the gazebo model to connect to sim_vehicle and everything works as expected. This is not the case with the plane, which leads me to believe it is something wrong with my gazebo setup/interface. I place my plane models (attached) into the corresponding ardupilot gazebo models folder.

If I run a modified launch file (attached) with my gazebo models --verbose, I get the following errors: 

```
ubuntu@ubuntu:~$ gazebo --verbose plane_ardupilot_demo.world
Gazebo multi-robot simulator, version 9.19.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 9.19.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.0.188
[Msg] Loading world file [/home/modalai/ardupilot_gazebo/worlds/plane_ardupilot_demo.world]
[Err] [Plugin.hh:178] Failed to load plugin libgazebo_airspeed_plugin.so: libgazebo_airspeed_plugin.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:178] Failed to load plugin libgazebo_gps_plugin.so: libgazebo_gps_plugin.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:178] Failed to load plugin libgazebo_motor_model.so: libgazebo_motor_model.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:178] Failed to load plugin libgazebo_imu_plugin.so: libgazebo_imu_plugin.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:178] Failed to load plugin libgazebo_groundtruth_plugin.so: libgazebo_groundtruth_plugin.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:178] Failed to load plugin libgazebo_magnetometer_plugin.so: libgazebo_magnetometer_plugin.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:178] Failed to load plugin libgazebo_barometer_plugin.so: libgazebo_barometer_plugin.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:178] Failed to load plugin libgazebo_mavlink_interface.so: libgazebo_mavlink_interface.so: cannot open shared object file: No such file or directory
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.0.188
[Err] [Plugin.hh:178] Failed to load plugin libForceVisual.so: libForceVisual.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:178] Failed to load plugin libForceVisual.so: libForceVisual.so: cannot open shared object file: No such file or directory
```

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Setup**
Ubuntu 18.04 on an ASUS laptop 

Thank you for your help! 
[plane_ardupilot_demo(world).txt](https://github.com/ArduPilot/ardupilot/files/7128988/plane_ardupilot_demo.world.txt)
[plane_models.zip](https://github.com/ArduPilot/ardupilot/files/7128996/plane_models.zip)

srmainwaring, 2022-01-23:
Hi @RotorBoy2020, there's been some work to migrate the ArduPilot plugin for Gazebo to support [Ignition Gazebo](https://ignitionrobotics.org/home) and provide a selection of updated models.

- the plugin is now located here: https://github.com/ArduPilot/ardupilot_gazebo
- there is more information on installation here: https://github.com/ArduPilot/ardupilot_gazebo/wiki
- a flying wing model ([Zephyr](https://github.com/ArduPilot/ardupilot_gazebo/tree/ignition-edifice/models/zephyr_with_ardupilot)) is included in the plugin repo
- a wider selection of models is being added here: https://github.com/ArduPilot/SITL_Models (see the models in the Ignition folder)

The model you obtained from the PX4 folder will not work without considerable modification. The plugins that are not loading are all particular to PX4-SITL and will not work with ArduPilot. The ArduPilot plugin uses a different protocol to communicate with it's version of SITL and applies the control commands it received from the flight controller differently (rather than use another plugin for motors or thrusters it commands the joint positions or velocities using PID controllers, rotor lift is obtained by using a physics plugin to set the lift-drag behaviour of the moving rotor link).

If you have any issues getting the simulation running you could ask for help on [ArduPilot Discourse]( https://discuss.ardupilot.org/), or if there is an issue with the plugin itself, here: https://github.com/ArduPilot/ardupilot_gazebo/issues

cheslee-z, 2022-12-17:
the ardupilot-gazebo plugin is semi-finished product. some models are ported from px4 and are not compatible with ardupilot.