ArduPilot/ardupilot #18623: Plane - CRSF glitching telemetry with Tracer

crashmatt, 2021-09-09:
**Issue details**
CRSF telemetry runs and stops with approx 5 second periods each.
This can be seem on the opentx transmitter telemetry and using a logic analyzer at the Ardupilot uart port. 
Appears to only be a problem when the GPS is active

CRSF is on SERIAL2 / TRX2 since the normal RCIN is on SERIAL7/TRX6 which does not have DMA
Mavlink telemetry on SERIAL1 is disconnected.

**Version**
Plane V4_1

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
Matek H743 Mini

**Logs**
_Please provide a link to any relevant logs that show the issue_

andyp1per, 2021-09-10:
Are you able to try the latest dev release, there are a couple of fixes that may resolve this for you. In particular DMA contention between CRSF and GPS is known.

crashmatt, 2021-09-10:
Tested both the 4.1 release and a build from 21-08-10.  Not sure if this is a dev branch or maintenance of some sort.
Will test the latest dev branch.

Thanks for your attention.

/Matt

crashmatt, 2021-09-10:
Built the latest master.   Now crsf does not work at all.
Building H743, not H743-dshot

andyp1per, 2021-09-10:
What RX do you have and what firmware version on it?

crashmatt, 2021-09-10:
Tracer Nano. V6.09.
Have just run the logic analyzer again and got nothing at all. The Rx should be sending crsf.

Why the update to ardupilot has anything to do with the Nano Rx setting is a mystery.  I disconnect the rx to do the update. 
 The Nano Rx is still set to crsf on that port.

I had a support ticket open with TBS on this topic.  I changed direction on debugging after putting a logic analyzer on the Ardupilot crsf stream and finding it was the source of the problems.

andyp1per, 2021-09-10:
There is a known issue with diversity RX's and CRSFv3 which they are working on. Also make sure you turn your TX on first - the negotiation for v3 fails if you start the RX first. I think I might be on 6.07 so this is possibly a new problem - I will have to try with my tracer setup (all working fine on 6.07)

crashmatt, 2021-09-10:
Same pattern as before.  You can see the changing pattern on the analyzer where ardupilot starts and stops telemetry.

[image]

crashmatt, 2021-09-10:
Will switch back to 6.07

crashmatt, 2021-09-10:
Same thing on 6.07
[image]

andyp1per, 2021-09-10:
Are you using Yaapu passthru? If so can you turn it off and see if that makes a difference.

andyp1per, 2021-09-10:
What serial port is the GPS connected on? What GPS do you have?

crashmatt, 2021-09-10:
Yaapu passthrough.  RC_OPTIONS=32 = No passthrough

GPS is Matek uBlox M8Q.
http://www.mateksys.com/?portfolio=m8q-5883

Moved CRSF to SERIAL6/TRX4.  Same glitchy result.

SERIAL0 USB MAVLINK2
SERIAL1 TRX7 MAVLINK2
SERIAL2 TRX1 None
SERIAL3 TRX2 GPS
SERIAL4 TRX3 No breakout
SERIAL5 RX8 SLCAN
SERIAL6 TRX4 RIN CRSF
SERIAL7 TRX6 (No DMA) - Matek recommended RIN

I wonder what am I doing wrong?

/Matt

crashmatt, 2021-09-10:
Console reports:  ArduPlane 4.2.0dev (0470fb2c)
This matches the git SHA1.

andyp1per, 2021-09-10:
There is a known problem with DMA sharing and high baudrate protocols. My suspicion is that this is what is happening. Pretty much all of the UARTs on that board share a DMA channel for transmit. As a test you could try saying:

DMA_NOSHARE UART2* UART6* 

in hwdef and see if that cures the problem. It will cause other things to not get DMA - so don't fly - but might tell us if this is in fact the issue.

andyp1per, 2021-09-10:
I setup a H743 slim with CRSF on serial 3 and GPS on serial 7 - all worked fine. This may be because serial 7 is not DMA enabled, I'll solder up another uart tomorrow to test.

andyp1per, 2021-09-10:
Are you using CAN? Try disconnecting it to see if that helps. In fact, if you can find a combination of disconnecting uarts w/o disconnecting CRSF and GPS that fixes the problem that will really help narrow things down.

crashmatt, 2021-09-11:
I can give that combination a go with a bit of rewiring.  I have quick connectors on all uarts but for the GPS.  Please let me test changing my hardware before you go hacking yours.

I see a "NO_DMA" tag with the SERIAL7 description.  Can this tag be applied to another UART to remove DMA support?  Might be easier than re-wiring.

andyp1per, 2021-09-11:
> I see a "NO_DMA" tag with the SERIAL7 description. Can this tag be applied to another UART to remove DMA support? Might be easier than re-wiring.

Yes it can

crashmatt, 2021-09-11:
I wonder if power changes are triggering this.  Do you have dynamic power set?

I have the opentx python script for crsf working on the logic analyzer output data. Have yet to decode all the information for any telltales of state change.

Not sure this is relevant - What transmitter do you use?
Mavlink is not connected in these tests.  Might that be causing some form of reset?

Have yet to try NO_DMA

andyp1per, 2021-09-11:
I have TX16S - no idea what might be triggering that’s what we’ve got to figure out!

crashmatt, 2021-09-11:
I have TX16s also.

Might have been tricking myself on the GPS being a problem.  Just unplugged it and still get the glitch.
Set all uarts except mavlink and crsf on trx4 to NO_DMA.

Will try moving the GPS uart next.

You have a J743-SLIM and I have a H743-MINI.  Difficult to know of any critical differences on these closed source boards.
Maybe I should obtain a SLIM board to compare against.

Can you share your ardupilot build?

andyp1per, 2021-09-11:
I have a mini as well. If you are able to reproduce on the mini with only the RX attached then seems like I should be able to reproduce with your settings. This is not a build - I have a whole bunch of FC's and RX's for development.

crashmatt, 2021-09-12:
Sorry, wrong terminology.  Can you share your firmware image?

My target airframe space is limited to 28mm wide.  The MINI only just fits in and I will need to delete the power connector.   I don't know of any other valid options.

andyp1per, 2021-09-12:
Can you post your full parameter list first - let me check with the firmware you are using

crashmatt, 2021-09-12:
Params are from QGC.  If you need a different format let me know.
[Dropbox link to Tracer_CRSF_Glitch_Ardupilot_RIN_SERIAL6-TRX4.params](https://www.dropbox.com/s/ntrd8uptgh1z67z/Tracer_CRSF_Glitch_Ardupilot_RIN_SERIAL6-TRX4.params?dl=0)
[Google drive link to Tracer_CRSF_Glitch_Ardupilot_RIN_SERIAL6-TRX4.params](https://drive.google.com/file/d/1Cc1D4GuRxwWXGdl8Vch29pP53lFPSFXN/view?usp=sharing)
The only change from default should be the RCIN UART.  I also switched off SERIAL8 in this case.

You might as well use the V4_1 release since that shows the problem.

I have the Rx running from 5V and GPS running from 4V5.  That enables me to keep them quiet in bootloader mode.  Read somewhere that might be a problem.  Have not yet checked the power supplies are ok.

crashmatt, 2021-09-12:
Didn't find an obvious problem on the power supplies.

Took a closer look at the crsf logic analyzer output.  The link statistics are the main data flow from the rx that may influence ardupilot. I don't see a significant change in data (tx power, antenna etc..) to trigger ardupilot to do change state.

crashmatt, 2021-09-13:
The firmware:
[Ardupilot_4_2_0dev_Tracer_CRSF_Glitch.bin.zip](https://drive.google.com/file/d/1gYfnCSBRp7KbJCKYNnKU61MGwl-uHH19/view?usp=sharing)

crashmatt, 2021-09-13:
One more point.  Ardupilot has no sensor calibrations, only CRSF port configuration.

crashmatt, 2021-09-13:
At the start of good telemetry there is a burst of flight mode, gps and battery data:

First value is timestamp.  Second value is milliseconds since the last time that telemetry was parsed.   We can see a 5s pause in  Attitude

[26.9373] [  4.0] [Radio]
[26.9413] [  4.0] [Radio]
[26.9419] [5069.4] [Attitude] pitch=62.791 roll=36.205 yaw=64.623
[26.9453] [  4.0] [Radio]
[26.9464] [5496.7] [Flight Mode] "RTL 

and this is normal with 0.5s interval:

crashmatt, 2021-09-13:
and this is normal gps/batt/with 0.5s interval:

[27.4415] [  4.0] [Radio]
[27.4455] [  4.0] [Radio]
[27.4466] [500.2] [Flight Mode] "RTL 
[27.4495] [  4.0] [Radio]
[27.4507] [500.2] [GPS] lat:59.548569 long:18.055079 speed:0 heading:0 alt:41 numsat:12
[27.4535] [  4.0] [Radio]
[27.4547] [500.1] [Battery] 7.7V 21.6A 355mAh
[27.4575] [  4.0] [Radio]

crashmatt, 2021-09-13:
Might have found an indicator of a problem.  Here is a ping devices message which appears to be at the end of the good telemetry:

[21.9322] [  4.0] [Radio]
[21.9362] [  4.0] [Radio]
[21.9365] [9998.2] [Ping Devices]
[21.9390] [9999.1] [Device] 0xc8 "ArduPlane V4.1.0beta5
[21.9402] [  4.0] [Radio]
[21.9443] [  4.0] [Radio]
[21.9483] [  4.0] [Radio]
[21.9497465] [Unknown] c8 03 0b c8 1e
[21.9523] [  4.0] [Radio]

Looking back at 10s earlier and removing the radio telemetry clearly shows the change of state around this message.

[11.5194] [120.5] [Attitude] pitch=62.800 roll=36.268 yaw=64.723
[11.5526] [100.1] [Link Statistics] ant1_rssi:40 ant2_rssi:0 uplink_status:100 uplink_snr:85 active_antenna:0 rf_mode:1 uplink_tx_power:2 downlink_rssi:29 downlink_status:100 downlink_dnr:98
[11.6390] [119.6] [Attitude] pitch=62.801 roll=36.269 yaw=64.720
[11.6528] [100.1] [Link Statistics] ant1_rssi:40 ant2_rssi:0 uplink_status:100 uplink_snr:84 active_antenna:0 rf_mode:1 uplink_tx_power:2 downlink_rssi:29 downlink_status:100 downlink_dnr:98
[11.7529] [100.1] [Link Statistics] ant1_rssi:44 ant2_rssi:0 uplink_status:100 uplink_snr:83 active_antenna:0 rf_mode:1 uplink_tx_power:2 downlink_rssi:29 downlink_status:100 downlink_dnr:98
[11.7591] [120.1] [Attitude] pitch=62.801 roll=36.269 yaw=64.716
[11.8530] [100.1] [Link Statistics] ant1_rssi:44 ant2_rssi:0 uplink_status:100 uplink_snr:82 active_antenna:0 rf_mode:1 uplink_tx_power:2 downlink_rssi:29 downlink_status:100 downlink_dnr:98
[11.8789] [119.8] [Attitude] pitch=62.801 roll=36.268 yaw=64.712
[11.9384] [10001.9] [Ping Devices]
[11.9399] [10000.7] [Device] 0xc8 "ArduPlane V4.1.0beta5
[11.9531] [100.1] [Link Statistics] ant1_rssi:41 ant2_rssi:0 uplink_status:100 uplink_snr:84 active_antenna:0 rf_mode:1 uplink_tx_power:2 downlink_rssi:29 downlink_status:100 downlink_dnr:98
[12.0533] [100.1] [Link Statistics] ant1_rssi:52 ant2_rssi:0 uplink_status:100 uplink_snr:78 active_antenna:0 rf_mode:1 uplink_tx_power:2 downlink_rssi:29 downlink_status:100 downlink_dnr:98
[12.1545] [101.2] [Link Statistics] ant1_rssi:49 ant2_rssi:0 uplink_status:100 uplink_snr:82 active_antenna:0 rf_mode:1 uplink_tx_power:2 downlink_rssi:29 downlink_status:100 downlink_dnr:98
[12.2547] [100.2] [Link Statistics] ant1_rssi:43 ant2_rssi:0 uplink_status:100 uplink_snr:80 active_antenna:0 rf_mode:1 uplink_tx_power:2 downlink_rssi:29 downlink_status:100 downlink_dnr:98

andyp1per, 2021-09-13:
Do you know where the ping is coming from? Is it the RX? There is another performance problem I am tracking with the TX CRSF menu that TBS are going to look into. It might be related to pings.

crashmatt, 2021-09-14:
Now with a tag showing which direction it is coming from.
Rx initiates with PIng Devices.
AP responds

There are some heartbeat message hidden.  Need to parse those to see if they have and hidden secrets.

[ AP] [11.8789] [119.8] [Attitude] pitch=62.801 roll=36.268 yaw=64.712
[ Rx] [11.9384] [10001.9] [Ping Devices]
[ AP] [11.9399] [10000.7] [Device] 0xc8 "ArduPlane V4.1.0beta5

crashmatt, 2021-09-14:
During normal telemetry there are no heartbeats from ardupilot:

[ Rx] [20.9702] [100.1] [Link Statistics] ant1_rssi:42 ant2_rssi:0 uplink_status:100 uplink_snr:85 active_antenna:0 rf_mode:1 uplink_tx_power:2 downlink_rssi:27 downlink_status:100 downlink_dnr:99
[ AP] [21.0289] [123.7] [Attitude] pitch=62.783 roll=36.213 yaw=64.687
[ Rx] [21.0703] [100.1] [Link Statistics] ant1_rssi:41 ant2_rssi:0 uplink_status:100 uplink_snr:86 active_antenna:0 rf_mode:1 uplink_tx_power:2 downlink_rssi:27 downlink_status:100 downlink_dnr:99
[ AP] [21.1494] [120.6] [Attitude] pitch=62.783 roll=36.212 yaw=64.686
[ Rx] [21.1704] [100.1] [Link Statistics] ant1_rssi:46 ant2_rssi:0 uplink_status:100 uplink_snr:85 active_antenna:0 rf_mode:1 uplink_tx_power:2 downlink_rssi:28 downlink_status:100 downlink_dnr:99
[ AP] [21.2692] [119.8] [Attitude] pitch=62.783 roll=36.212 yaw=64.686
[ Rx] [21.2705] [100.1] [Link Statistics] ant1_rssi:38 ant2_rssi:0 uplink_status:100 uplink_snr:87 active_antenna:0 rf_mode:1 uplink_tx_power:2 downlink_rssi:28 downlink_status:100 downlink_dnr:99
[ Rx] [21.3707] [100.1] [Link Statistics] ant1_rssi:43 ant2_rssi:0 uplink_status:100 uplink_snr:84 active_antenna:0 rf_mode:1 uplink_tx_power:2 downlink_rssi:28 downlink_status:100 downlink_dnr:99
[ AP] [21.3894] [120.2] [Attitude] pitch=62.784 roll=36.212 yaw=64.688
[ AP] [21.4497] [500.2] [Flight Mode] "RTL 

During the telemetry glitch there are heartbeats:

[ Rx] [21.9715] [100.1] [Link Statistics] ant1_rssi:39 ant2_rssi:0 uplink_status:100 uplink_snr:86 active_antenna:0 rf_mode:1 uplink_tx_power:2 downlink_rssi:28 downlink_status:100 downlink_dnr:100
[ Rx] [22.0716] [100.1] [Link Statistics] ant1_rssi:41 ant2_rssi:0 uplink_status:100 uplink_snr:85 active_antenna:0 rf_mode:1 uplink_tx_power:2 downlink_rssi:28 downlink_status:100 downlink_dnr:100
[ AP] [22.1496] [199.9] [Heartbeat] Device:0xc8
[ Rx] [22.1727] [101.1] [Link Statistics] ant1_rssi:46 ant2_rssi:0 uplink_status:100 uplink_snr:84 active_antenna:0 rf_mode:1 uplink_tx_power:2 downlink_rssi:28 downlink_status:100 downlink_dnr:100
[ Rx] [22.2729] [100.2] [Link Statistics] ant1_rssi:38 ant2_rssi:0 uplink_status:100 uplink_snr:82 active_antenna:0 rf_mode:1 uplink_tx_power:2 downlink_rssi:28 downlink_status:100 downlink_dnr:100
[ AP] [22.3496] [200.0] [Heartbeat] Device:0xc8
[ Rx] [22.3730] [100.1] [Link Statistics] ant1_rssi:40 ant2_rssi:0 uplink_status:100 uplink_snr:85 active_antenna:0 rf_mode:1 uplink_tx_power:2 downlink_rssi:28 downlink_status:100 downlink_dnr:100
[ Rx] [22.4731] [100.1] [Link Statistics] ant1_rssi:40 ant2_rssi:0 uplink_status:100 uplink_snr:85 active_antenna:0 rf_mode:1 uplink_tx_power:2 downlink_rssi:28 downlink_status:100 downlink_dnr:100
[ AP] [22.5532] [203.6] [Heartbeat] Device:0xc8

crashmatt, 2021-09-14:
Starting to make sense.

https://github.com/ArduPilot/ardupilot/blob/221e7c612e0d282f08f391decbbdc4583bb39f32/libraries/AP_RCTelemetry/AP_CRSF_Telem.cpp#L245

crashmatt, 2021-09-14:
Yes:
https://github.com/ArduPilot/ardupilot/blob/221e7c612e0d282f08f391decbbdc4583bb39f32/libraries/AP_RCTelemetry/AP_CRSF_Telem.cpp#L274

crashmatt, 2021-09-14:
It is going into params mode since it is doing the PingDevices message.  It is not in the CRSF Tx menu.
Tested shortening the heartbeat period to 1s.  That did not mute the PingDevices message.

Filtering PingDevices out of params mode appears to work:

`    if (!_custom_telem.params_mode_active && _pending_request.frame_type > 0 && _pending_request.frame_type != AP_RCProtocol_CRSF::CRSF_FRAMETYPE_PARAM_DEVICE_INFO) {
        // fast window start
        _custom_telem.params_mode_start_ms = now_ms;
        _custom_telem.params_mode_active = true;
        enter_scheduler_params_mode();
    } else if ( (expired || _pending_request.frame_type==0) && _custom_telem.params_mode_active) {
        // fast window stop
        _custom_telem.params_mode_active = false;
        exit_scheduler_params_mode();
    }`

Ardupilot parameter collection seems to be normal.
I need to take another logic analyzer measurement with this setup.  Maybe take a look at the parameter collection phase while I am at it.

andyp1per, 2021-09-14:
Yes you are right, this is because of the enablement of parameter passthrough support. The bug seems to be that a ping request can be sent ad-hoc and this is what is removing the telemetry. I'll discuss with @yaapu the best way to fix this.

crashmatt, 2021-09-14:
Why doesn't your system do the ad-hoc device ping?  That or why does it not respond the same way? How is your setup different?

Might be good to only enable parameter download in safe mode to prevent long timeouts in flight mode.

andyp1per, 2021-09-14:
Yeah not sure, seems strange. Do you have a TBS VTX attached as well?

crashmatt, 2021-09-14:
I don't have VTX.
Tracer Tx setting for VTX Control is set to Rx.  I don't remember changing this from factory default.

Maybe I can get mavlink working properly now.

andyp1per, 2021-09-14:
The system I am testing on has VTX as well - I wonder if that makes a difference

andyp1per, 2021-09-15:
Can you try https://github.com/ArduPilot/ardupilot/pull/18672 to see if it helps?

crashmatt, 2021-09-15:
Got mavlink working.  Parameter download is pretty slow and unreliable.  Might be necessary to add a limited telemetry rate mode for mavlink parameters.  I don't yet know if limiting crsf telemetry will improve mavlink rate.

IamPete1, 2022-09-04:
Any updates? Is this resolved?

crashmatt, 2022-09-12:
Has been a long while since I looked at this.  Got stuck waiting for TBS to
offer higher bandwidth telemetry profiles.  Close it for now.

closed 2022-09-14 by IamPete1