ArduPilot/ardupilot #18628: GPS does not report status until locked

lthall, 2021-09-10:
GPS does not report the number of satellites until it has achieved a lock. This means we are not reporting the correct status of the GPS.

This give the impression that the GPS has been detected but is not working.

It would be nice if we could see the GPS is healthy before it has a lock.

IamPete1, 2021-09-10:
I don't see this in the code, it might be a GCS thing.

IamPete1, 2021-09-19:
@lthall Can you provide some more info, it could be down to the particular GPS type. I guess you were using mission planner?

lthall, 2021-09-29:
I tested this with HERE, Here 2, and Z-F9P. I connected the F9P to U-center so I could see the satellites on U-Centre but we don't see any satellites on mission planner or the Mavlink Interceptor until we have a fix. On both Here and Here 2 we see an instantaneous increase in sat count from zero at the same time as we get a fix.

Maybe the message we take our sat count is only provided after a fix. U-Centre may be getting the sat count from a different message that is present when we don't have a fix.

Sorry for the delay in my reply I needed to sit down and test this properly to answer your question.

When I notice this is setting up my gps for the first time. Because they are getting a cold boot it can take a while to get a fix even through they have satellites. When looking at mission planner it looks like the gps is broken but u-centre shows the gps is working and has plenty of sats, just working on that first fix. 

My concern is this could take long enough in the field for users to think there are problems and start trying to power cycle when if they could see the sat count they would realize it was just working on the first lock.

WickedShell, 2021-09-29:
Note that depending on where you look in u center you may be mislead. There
is a difference between satellites tracked, and satellites used, and in
ardupilot we only care about the latter. I wouldn't be surprised if that's
what you are seeing. 0 satellites are used until there is a fix, at which
point it jumps to immediately. This sounds like it's working as intended.

The thing to help the UX of the GCS would be to either convey both numbers,
or the signal to noise ratios, but we lack the MAVLink bandwidth for the
latter normally...

lthall, 2021-09-29:
>>  0 satellites are used until there is a fix, at which point it jumps to immediately. 

Ok, that confirms this is not a bug. Thanks. I will close this issue.

closed 2021-09-29 by lthall