ArduPilot/ardupilot #18632: --console --map not working right after build-up with the official instructions...

tychien, 2021-09-10:
Hi, I'm working on Ubuntu 18.04 LTS 
I'm doing the simulator on my x86 computer. 
It worked fine with the console and map popped up on my old computer (which is also 18.04 LTS) . 
The Ardupilot worked fine with Gazebo and it could received the command from the terminal, but there's no console and map windows showing up after I type sim_vehicle.py ---console --map in ardupilot/ArduCopter directory in the terminal. 
Only xterm shown up like as below. 
How can I fix that? 

[image]

IamPete1, 2021-09-10:
Do they load if you type `module load map` or `module load console` in MAVProxy?

tychien, 2021-09-10:
it returns with errors:

for `module load map` :
module load map
STABILIZE> Process Process-4:
Traceback (most recent call last):
  File "/usr/lib/python3.6/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib/python3.6/multiprocessing/process.py", line 93, in run
    self._target(*self._args, **self._kwargs)
  File "/home/tychien/.local/lib/python3.6/site-packages/MAVProxy/modules/mavproxy_map/mp_slipmap.py", line 98, in child_task
    self.app.frame = MPSlipMapFrame(state=self)
  File "/home/tychien/.local/lib/python3.6/site-packages/MAVProxy/modules/mavproxy_map/mp_slipmap_ui.py", line 48, in __init__
    self.SetIcon(icon.SimpleIcon("MAP").get_ico())
  File "/home/tychien/.local/lib/python3.6/site-packages/MAVProxy/modules/lib/icon.py", line 73, in __init__
    fig.savefig(buf, format='png')
  File "/usr/lib/python3/dist-packages/matplotlib/figure.py", line 1834, in savefig
    self.canvas.print_figure(fname, **kwargs)
AttributeError: 'NoneType' object has no attribute 'print_figure'
ERROR in command ['load', 'map']: map not ready
Traceback (most recent call last):
  File "/home/tychien/.local/bin/mavproxy.py", line 747, in process_stdin
    fn(args[1:])
  File "/home/tychien/.local/bin/mavproxy.py", line 559, in cmd_module
    load_module(modname, **kwargs)
  File "/home/tychien/.local/bin/mavproxy.py", line 498, in load_module
    module = m.init(mpstate, **kwargs)
  File "/home/tychien/.local/lib/python3.6/site-packages/MAVProxy/modules/mavproxy_map/__init__.py", line 876, in init
    return MapModule(mpstate)
  File "/home/tychien/.local/lib/python3.6/site-packages/MAVProxy/modules/mavproxy_map/__init__.py", line 69, in __init__
    self.map = mp_slipmap.MPSlipMap(service=service, elevation=True, title=title)
  File "/home/tychien/.local/lib/python3.6/site-packages/MAVProxy/modules/mavproxy_map/mp_slipmap.py", line 75, in __init__
    raise Exception("map not ready")
Exception: map not ready

for `module load console` : 
module load console
STABILIZE> ERROR in command ['load', 'console']: 'NoneType' object has no attribute 'print_figure'
Traceback (most recent call last):
  File "/home/tychien/.local/bin/mavproxy.py", line 747, in process_stdin
    fn(args[1:])
  File "/home/tychien/.local/bin/mavproxy.py", line 559, in cmd_module
    load_module(modname, **kwargs)
  File "/home/tychien/.local/bin/mavproxy.py", line 498, in load_module
    module = m.init(mpstate, **kwargs)
  File "/home/tychien/.local/lib/python3.6/site-packages/MAVProxy/modules/mavproxy_console.py", line 657, in init
    return ConsoleModule(mpstate)
  File "/home/tychien/.local/lib/python3.6/site-packages/MAVProxy/modules/mavproxy_console.py", line 41, in __init__
    mpstate.console = wxconsole.MessageConsole(title='Console', ico=icon.SimpleIcon("CONSOLE"))
  File "/home/tychien/.local/lib/python3.6/site-packages/MAVProxy/modules/lib/icon.py", line 73, in __init__
    fig.savefig(buf, format='png')
  File "/usr/lib/python3/dist-packages/matplotlib/figure.py", line 1834, in savefig
    self.canvas.print_figure(fname, **kwargs)
AttributeError: 'NoneType' object has no attribute 'print_figure'

IamPete1, 2021-09-10:
Certainly sounds like a python library issue, I'm not sure exactly what you need to do. You could try setting `moddebug 3` and see if there is a more useful error message.

stephendade, 2021-09-11:
This appears to be an issue related to @khancyr's recent icon additions to MAVproxy.

Can you do a ``pip3 list`` so we can check your package versions?

tychien, 2021-09-11:
@IamPete1  that's all the error messages  shown above after `set moddebug 3` then `module load map`

@stephendade  there you go

Package                       Version
----------------------------- -------------------
actionlib                     1.12.1
angles                        1.9.12
apturl                        0.5.2
asn1crypto                    0.24.0
beautifulsoup4                4.6.0
bondpy                        1.8.5
Brlapi                        0.6.6
camera_calibration            1.15.0
camera_calibration_parsers    1.11.13
catkin                        0.7.29
certifi                       2018.1.18
chardet                       3.0.4
click                         6.7
colorama                      0.3.7
command-not-found             0.3
controller_manager            0.18.4
controller_manager_msgs       0.18.4
cryptography                  2.1.4
cupshelpers                   1.0
cv_bridge                     1.13.0
cycler                        0.10.0
defer                         1.0.6
diagnostic_analysis           1.9.7
diagnostic_common_diagnostics 1.9.7
diagnostic_updater            1.9.7
distro-info                   0.18ubuntu0.18.04.1
dronekit                      2.9.2
dronekit-sitl                 3.3.0
dynamic_reconfigure           1.6.3
future                        0.18.2
gazebo_plugins                2.8.7
gazebo_ros                    2.8.7
gencpp                        0.6.5
geneus                        2.2.6
genlisp                       0.4.16
genmsg                        0.5.16
gennodejs                     2.0.1
genpy                         0.6.16
html5lib                      0.999999999
httplib2                      0.9.2
idna                          2.6
image_geometry                1.13.0
interactive-markers           1.11.5
joint_state_publisher         1.12.15
kazam                         1.4.5
kdl_parser_py                 1.13.1
keyring                       10.6.0
keyrings.alt                  3.0
language-selector             0.1
laser_geometry                1.6.7
launchpadlib                  1.10.6
lazr.restfulclient            0.13.5
lazr.uri                      1.0.3
linecache2                    1.0.0
louis                         3.5.0
lxml                          4.2.1
macaroonbakery                1.1.3
Mako                          1.0.7
MarkupSafe                    1.0
matplotlib                    2.1.1
MAVProxy                      1.8.41
mavros                        1.8.0
message_filters               1.14.11
monotonic                     1.6
[... 102 lines cut ...]
systemd-python                234
testresources                 2.0.0
tf                            1.12.1
tf_conversions                1.12.1
tf2_geometry_msgs             0.6.5
tf2_kdl                       0.6.5
tf2_py                        0.6.5
tf2_ros                       0.6.5
topic_tools                   1.14.11
traceback2                    1.4.0
ubuntu-advantage-tools        27.2
ubuntu-drivers-common         0.0.0
ufw                           0.36
unattended-upgrades           0.1
unittest2                     1.1.0
urdfdom-py                    0.4.5
urllib3                       1.22
usb-creator                   0.3.3
wadllib                       1.3.2
webencodings                  0.5
wheel                         0.30.0
wxPython                      4.0.1
xacro                         1.13.12
xkit                          0.0.0
zope.interface                4.3.2

stephendade, 2021-09-11:
hmm, you have some old packages there. Try upgrading:
``pip install -U matplotlib wxPython --user``

tychien, 2021-09-11:
It keeps showing the error messages.  But then I tried to upgrade all the packages I have: 

`pip list --outdated --format=freeze | grep -v '^\-e' | cut -d = -f 1  | xargs -n1 pip install -U` 

And it worked!  I think it's something to do with the outdated packages...
Thank you so much!  I appreciate your patience and assistance!!!!!

closed 2021-09-11 by tychien