ArduPilot/ardupilot #18634: Plane recovered in inverted flight after a stall spin, FBWA flight mode - slow roll

dramida, 2021-09-10:
I have a Believer 1960 twin motor, well tuned. After a **induced stall** in FBWA, **it recovered in inverted flight** and flew like this plunging to earth until I forcefully rolled right 180 degrees. Plane obeyed **roll command slowly**, it flew for a while in knife edge.
Happily it was 350m AGL when I start stall and recovered at 80m.
20 seconds of terror.
[image]

Flight was 41 min long, the issue is clearly visible at 18:47:10
Logs on the drop box here:
https://www.dropbox.com/sh/z31fwh8sgw4j2aq/AABPlImAwZGUy8e47La_RIefa?dl=0

IamPete1, 2021-09-10:
AP was giving full aileron and full elevator, not very much more that it can do. 

[image]

IamPete1, 2021-09-12:
I don't think thinks is a bug, so I'm going to close, the forum is the the best place for advice on such things. https://discuss.ardupilot.org/

closed 2021-09-12 by IamPete1