ArduPilot/ardupilot #18681: Plane throttle output noisy when flying with minimum groundspeed

peterbarker, 2021-09-16:
## Bug report

**Issue details**

Plane's throttle output looks like this when running a minimum groundspeed:

[image]

groundspeed undershoot (https://github.com/ArduPilot/ardupilot/pull/18679) looks like this:
[image]

target airspeed looks like this:
[image]

Most interesting graph is sp/spdem  and dsp/dspdem on the same graph:

[image]

mission can be extracted from logfile below, but is the relatively standard CMAC-circuit.txt with three `DO_CHANGE_SPEED`s added to it.

```
QGC WPL 110
0	0	0	16	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
1	0	3	22	15.000000	0.000000	0.000000	0.000000	-35.359833	149.164703	41.029999	1
2	0	3	16	0.000000	0.000000	0.000000	0.000000	-35.359585	149.161392	100.000000	1
3	0	0	178	1.000000	15.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
4	0	3	16	0.000000	0.000000	0.000000	0.000000	-35.366463	149.162231	100.000000	1
5	0	0	178	1.000000	16.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
6	0	3	16	0.000000	0.000000	0.000000	0.000000	-35.366131	149.164581	100.000000	1
7	0	0	178	1.000000	17.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
8	0	3	16	0.000000	0.000000	0.000000	0.000000	-35.359272	149.163757	100.000000	1
9	0	0	177	2.000000	-1.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
10	0	3	16	0.000000	0.000000	0.000000	0.000000	-35.359272	149.163757	100.000000	1
```

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
SITL default

**Hardware type**
SITL

**Logs**
[00000007.BIN.txt](https://github.com/ArduPilot/ardupilot/files/7174638/00000007.BIN.txt)

peterbarker, 2021-09-16:
For reference, the same mission flown with the `DO_CHANGE_SPEED` first parameter changed to `0` - airspeed:

[image]

vrquaeler, 2021-10-05:
Hi Peter,
I have many real world log files (foam plane, different versions of AP4.0.x on onmibus4pro without airspeed sensor) successfully flown in strong winds using MIN_GND_SPD_CM. No throttle pulsing there. This is still the "foam plane saver" for me.