ArduPilot/ardupilot #18687: Plane: suggest disable flap_slewrate during takeoff mode

jinchengde, 2021-09-17:
## Feature request

suggest disable flap_slewrate during takeoff mode

**Describe the solution you'd like**
if we choose automatic takeoff and set the flap in takeoff mode 20percent, and the FLAP_SLEWRATE set too small like 25, the the flap may not deploy complete after the plane had reach the rotate speed

**Describe alternatives you've considered**
disable flap_slewrate during takeoff
[https://github.com/jinchengde/ardupilot/blob/master/ArduPlane/servos.cpp#L593-L594](url)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

magicrub, 2021-09-17:
Overriding slew sounds like a bad idea. Your example doesn't make sense either. Why would you be switching modes so quickly just as you're hitting rotate speed?

jinchengde, 2021-09-18:
let me explain the details:
1, the first item in mission is takeoff
2, switch the mode to auto, plane start to speed up, meanwhile the flap start to deploy
3. the plane rotate speed(v2 speed) is 10m/s, but maybe 8m/s if with flap
4. set the rotate speed is 8m/s, but the flap may not deploy complete when the speed had reach 8m/s if we set the FLAP_SLEWRATE too small
normally, in real plane, pilot always deploy the flap 5 - 10 degree before they start to takeoff, should we consider that?

IamPete1, 2021-09-18:
Can you not switch into auto before arming, wait for flaps and then arm?

jinchengde, 2021-09-19:
actually that is not about arm, that is automatic flap in takeoff.
set TKOFF_FLAP_PCNT = 20, FLAP_SLEWRATE = 10
then if you command automatic runway takeoff, the flap start to deploy but very slow,  sometime could not deploy complete before plane lift-off.
if we set FLAP_SLEWRATE = 50, the automatic flap deploy maybe very quickly
but we always set LAND_FLAP_PERCNT = 100, if we have so quickly flap deploy, the plane will have a jump when start to approaching
so I suggest in automatic runway takeoff, force to set FLAP_SLEWRATE = 100, then we could have quickly flap deploy before vehicle speed up

IamPete1, 2021-09-19:
How are you commanding the takeoff? Are you taking off after a landing in a mission? Maybe you can detail the use case a little more. A example mission would be great.

I think just disabling flap slew-rate is a bad idea, but we could wait for to start the takeoff until the flaps have finished moving.

jinchengde, 2021-09-20:
> How are you commanding the takeoff? Are you taking off after a landing in a mission? Maybe you can detail the use case a little more. A example mission would be great.
switch the flight mode to auto, the first item in mission is takeoff
> How are you commanding the takeoff? Are you taking off after a landing in a mission? Maybe you can detail the use case a little more. A example mission would be great.
that is another way to solve

IamPete1, 2021-09-20:
> switch the flight mode to auto, the first item in mission is takeoff, and the flap position current is 0, so the flap start to deploy to the position of TKOFF_FLAP_PCNT
> 
> >

So what happens if you switch to auto first and then arm once the flaps are done moving?

peterbarker, 2024-11-24:
We have a comment in the code indicating this was wanted before 2016... 
```
            case AP_FixedWing::FlightStage::NORMAL:
                if (g.takeoff_flap_percent != 0 && in_preLaunch_flight_stage()) {
                    // TODO: move this to a new FLIGHT_PRE_TAKEOFF stage
                    auto_flap_percent = g.takeoff_flap_percent;
                }
                break;
```