ArduPilot/ardupilot #18688: Add Protection from Arming via MAVLink (e.g. Mission Planner) when throttle not at minimum

ktrussell, 2021-09-17:
PROBLEM: The autopilot can be armed from Mission Planner (and presumably any MAVLink source) when the throttle is not at the neutral position.

OBSERVATION: The RC_OPTIONS parameter can be set to prevent arming using the joystick method when the throttle is not in the neutral position, but the vehicle can still be armed from Mission Planner even if the throttle is not at neutral.  This can present a potentially dangerous situation.

POTENTIAL SOLUTIONS:
1. Perhaps the easiest solution is to make the arming from MAVLink obey the RC_OPTIONS setting. I cannot think of a case when it would be desirable to disallow arming from the joystick when the throttle is in a dangerous state but desirable to allow arming from Mission Planner. The only negative that I see to this solution is that the name of the parameter "RC_OPTIONS" implies that it only refers to RC not MAVLink.
2.  Assuming that the first option is not acceptable, then an additional parameter or an additional setting of an existing parameter could be created to allow selection of this feature.

ALTERNATIVES CONSIDERED:
A Lua Script can be written that will immediately disarm if it detects that the autopilot has been armed with the throttle away from neutral. In fact, Yuri_Rage has written such a script. The script is presented here: https://discuss.ardupilot.org/t/how-can-i-prevent-arming-from-mission-planner-if-throttle-is-not-at-0/75965/7
While this script does present some level of protection, it is not as ideal as a built-in feature would be because (1) the vehicle will be momentarily armed until the script catches it, (2) Lua scripts have low priority and are not guaranteed to run, and (3) Lua Scripts can be disabled.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[X] Rover
[  ] Submarine

IamPete1, 2021-09-18:
I cant replicate this in master, I get: `PreArm: Throttle (RC3) is not neutral`. Are you on 4.1 beta? Can you share your params and/or a log.

Thanks

peterbarker, 2021-09-20:
[image]

KimHyungSub, 2021-09-20:
Maybe, I guess that @ktrussell mentions "force arm" on the Mission Planner.

[image]

ktrussell, 2021-09-24:
Sorry for my tardiness. I do not get the Force Arm message shown above. 

If I have my throttle at full and click on the ARM/DISARM button on the ACTION tab in Mission Planner, ArduRover immediately arms and slams the throttle sticks full on. 

I just verified again to be sure I was not seeing things. Attached are my current parameters. I am running 4.1.0-RC3.
[2021-08-29 4-1-0rc1.zip](https://github.com/ArduPilot/ardupilot/files/7221813/2021-08-29.4-1-0rc1.zip)

IamPete1, 2021-09-24:
@ktrussell you have arming checks disabled. `ARMING_CHECK,0` 

Set `ARMING_CHECK` to 1 to re-enable all checks. 

It there is some VERY good reason why you do not want all checks you could just enable the RC checks.

IamPete1, 2021-09-24:
@rmackay9 We could add RC inputs to the mandatory checks? Unless were going to do that we might as well close this...

rmackay9, 2021-09-24:
@IamPete1 thanks very much for investigating this.  With arming checks off it makes sense that this could happen. 

@ktrussell, I'll close this if that's OK but thanks very much for the report and I'll be notified of responses if we need to review further.

closed 2021-09-24 by rmackay9

ktrussell, 2021-09-24:
I verified that with ARMING_CHECK set to 1 (ALL), ArduRover does indeed NOT let me arm if throttle is away from neutral. Then I tried setting it to 64, which is RC Channels, and that also does not let arming happen, so that is the setting I want. I should have just tried it instead of asking. I was lead to believe that RC Channels didn't do this, but I am happy it does. I apologize for taking up you guys' time, but thank you!

I will look closely at the arming checks. I may be OK with all of them being enabled.