ArduPilot/ardupilot #18715: CRSF not recognized after reboot with power applied

Hwurzburg, 2021-09-19:
MatekH743Wing, CRSF attached to UART6,ALT_BRD_CONFIG=1, CRSF powered from 4.5V

If powered by USB first or battery first, CRSF detected
If powered by Battery first and USB then connected, CRSF detected, but issuing a reboot, and CRSF is no detected

4.2 master as of yesterday
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover
[  ] Submarine

andyp1per, 2021-09-19:
This is down to the RX and the v3 protocol, nothing I can do about it.

Hwurzburg, 2021-09-19:
if it can be initialized at power up, why cant it be initialized on a reboot?

andyp1per, 2021-09-19:
The RX is not rebooted, only the flight controller.

Hwurzburg, 2021-09-19:
and no way the FC can reboot it also? is it locked up until a power cycle since it missed communicating with the FC? this means there is never a recovery after a WD in flight....whereas other RXs can, correct?

andyp1per, 2021-09-20:
> 
> 
> and no way the FC can reboot it also? is it locked up until a power cycle since it missed communicating with the FC? this means there is never a recovery after a WD in flight....whereas other RXs can, correct?

I don't know - you'd have to test it and see. Because of the baud rate negotiation I'm not sure what we would do on a WD - the RX will fallback to the slower baud rate eventually but that may be too late to save the aircraft

Hwurzburg, 2021-09-20:
maybe we should pick some brains on the devcall.....it is a real pain, since field reboots can happen often during testing....but unless the RX is irrevocably locked up, we should be able to re-init it on a reboot, I would think....

jaroszmm, 2022-01-10:
This also happens the other way round - my nanorx on 6.10 lost power in flight (probably bad crimp or solder), plane went to RTL and loitered till battery died. 
Downgraded from 4.2 dev to 4.1 official and all is fine.