ArduPilot/ardupilot #18730: AC 4.0.4-rc2 - Guided resets attitude/twitch on each Guided command

AndKe, 2021-09-21:
## Bug report
**Issue details**
The autopilot is flying in guided mode and a companion computer is sending new waypoints to the autopilot.
We expect, the drone to fly smoothly. But, we see that the drone stops for 50 ms when it recieves a new guided waypoint, even if the waypoint is identical to the previous one

Please see video:
https://drive.google.com/file/d/1fhcC1d3LTkbb-uDESppl27GcWyU0AdsM/view

**Version**
AC 4.0.4-rc2 

**Platform**
[  ] All
[  ] AntennaTracker
[x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
quad

**Hardware type**
SITL

**Logs**

rmackay9, 2021-09-21:
@AndKe,

Thanks for the report.  Could you try with Copter-4.1.0-rcX?  We've made so many changes to the position controller and guided mode in Copter-4.1 and it is unlikely that we will release any more updates to 4.0.x.

lthall, 2021-09-21:
And please provide a log.

AndKe, 2021-09-21:
Right now, I think I see a significant improvement on current master, will repeat the very same mission, and post log if this is still a issue.  - will keep you updated.

lthall, 2021-09-24:
Is this still an issue?

AndKe, 2021-09-24:
@lthall the researcher working on it had trouble using such new builds with Airsim . I requested a test with sim_vehicle only, in order to get an answer to this. I've been promised an result during the next week.