ArduPilot/ardupilot #18737: Unable to automatically create the dir: “APM / logs”

AwesomeOY, 2021-09-22:
## Bug report

**Clean all file contents in TF card and open log_ Disarm = 1. GCS receive message: "PreArm: Logging failed".
After manually creating this directory, all log data records are normal. **

**ArduCopter V4.2.0-dev (d853d16c)**

** Problem code **
void AP_Logger_File::ensure_log_directory_exists()
{
    int ret;
    struct stat st;

    EXPECT_DELAY_MS(3000);
    ret = AP::FS().stat(_log_directory, &st);
    if (ret == -1) {
        ret = AP::FS().mkdir(_log_directory);
    }
    if (ret == -1 && errno != EEXIST) {
        printf("Failed to create log directory %s : %s\n", _log_directory, strerror(errno));
    }
}
strerror(errno) is EIO.This error message indicates an IO error.

**Platform**
Copter

**Airframe type**
Quad

**Hardware type**
Pixhawk

IamPete1, 2021-09-22:
There are a bunch of related issues:
https://github.com/ArduPilot/ardupilot/issues/17257
https://github.com/ArduPilot/ardupilot/issues/16834
https://github.com/ArduPilot/ardupilot/issues/16103
https://github.com/ArduPilot/ardupilot/issues/14860

AwesomeOY, 2021-09-22:
@IamPete1 I have this problem every time. No reason has been found.

tridge, 2021-11-15:
fixed by #19267

closed 2021-11-15 by tridge

AwesomeOY, 2021-11-23:
> fixed by #19267

Okay!It's great!