ArduPilot/ardupilot #18750: Copter: likely FMU reboot in air and crashed down to ground with one motor spinned a long time before stopped

lukezhqin, 2021-09-23:
## Bug report

**I used my copter when I tried to modify some avoidance code to test filght. Suddenly all six mtors just stopped then it fell down to the ground. But one of the motors spinned fast for a long time before all stopped. Meanwhile  I found the hadware safety switch is just engaged and not armed.**

_I met this condition twice then, I'd like to ask for some help here. In which condition, whould the PWM sent out from IO but FMU is not armed and even the hadware safety switch is not yet been pressed? _

**Version**
_Copter V3.6.8_

**Platform**
[  ] All
[  ] AntennaTracker
[ × ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_hex_

**Hardware type**
_Pixhawk2_

**Logs**
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2021-09-23:
@lukezhqin, any chance of an onboard log.  TBH, 3.6.x is quite old now so there is really very little chance that we will release a fix but..

lukezhqin, 2021-09-23:
@rmackay9 , Thanks very much for reply.
I just don't understand where did the PWM come from. In that situation, I checked the LED is lighting and NTF is beep, which means a reboot on FMU. But one of the motors kept running for about 10 minuts before all stopped. The motor seemed like have kept a "ramp" type signal. I wonder if anyone met this question?

IamPete1, 2021-09-23:
It sounds like your running a custom version of Copter 3.6.8? In which case all bets are off. If you can reproduce on stock 4.0 or 4.1 then please reopen this issue. 

If you can provide a log and a link to your custom branch we will have a look and see if there is anything we can spot. The forum would be a better place for that.

closed 2021-09-23 by IamPete1