ArduPilot/ardupilot #18765: Setting Int32 params with bit>23 set leads to storing of mangled value

shellixyz, 2021-09-25:
## Bug report

**Issue details**

At least works with OSD_OPTIONS but I guess it applies to all Int32 params

`param set OSD_OPTIONS 1073741825` leads to 1073741824 being stored in AP_OSD.options

[image]

Same thing happens with any integer with one bit above the 23th bit set to 1: the first bit is stored at 0 even if it was 1.

**Version**
master

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

WickedShell, 2021-09-25:
This is well known, the param transport is a float, so you are limited to what can easily be stored in the mantissa with accuracy, which is what you are seeing There is a reason none of the bitmasks have more then 23 bits....

shellixyz, 2021-09-25:
Ok, still a bit strange not to have any error show up and just storing a value which is different from what has been specified though don't you think ?

WickedShell, 2021-09-26:
The value stored is what was sent. The GCS sent a float, so the autopilot
did what it was supposed to. Really the GCS should be responsible for
checking that out was what you requested, but depending upon the GCS
implementation it may or may not do that. FWIW my GCS would give you an
error here, but it wouldn't be able to force the value to what you wanted.
It's something to consider more as a MAVProxy issue here.

shellixyz, 2021-09-26:
@WickedShell  It seems like a Mavproxy issue then indeed. What is your GCS ? Is it publicly available ?