ArduPilot/ardupilot #18780: Copter4.1.0 run in Pixhawk, will reboot if batt_monitor changed, even in air!!! My drone crashed today!!

ranqingfa, 2021-09-27:
Hi all

Copter4.1.0 run in Pixhawk, will reboot if batt_monitor changed, even in air!!! 

**Version**
Copter 4.1.0-rc2

**Platform**
[  ] Copter

**Airframe type**
QuadOcta

**Hardware type**
Pixhawk

**Logs**

ranqingfa, 2021-09-27:
Seems ram full then reboot

IamPete1, 2021-09-27:
@ranqingfa Do you have a log for the flight? What did you change it too and from?

ranqingfa, 2021-09-29:
> @ranqingfa Do you have a log for the flight? What did you change it too and from?

The log just suddenly stop after pamameter changed, I think its useless. we change the batt_monitor from 4 to 8, try to use a uavcan battery

ranqingfa, 2021-09-29:
Actually autopilot system was dead, not reboot, after we re-power drone, works well with batt_monitor=8

rmackay9, 2021-10-12:
I tried to reproduce this and wasn't able to.  That doesn't mean that it's not possible to reproduce but I guess we need some more info on the steps.