ArduPilot/ardupilot #18802: DJI SBUS baud fast protocol support

Luix333, 2021-09-29:
Feature request

Hi, newbie here. I came this way from the ardupilot blog after being told this is the right place to ask people way smarter than me to add a new feature.

I noticed that, when using my DJI transmitter, I have to avoid using DJI's Fast Baud Protocol since it isn't currently supported on ardupilot. In betaflight it can be enable through "set sbus_baud_fast = ON", but as far as I could see, this is not supported currently on Ardupilot.

**Describe the solution you'd like**

It's pretty simple really (to ask). I want to request support for DJI HDL FAST BAUD SBUS protocol to be added to Ardupilot. 

According to someone on the blog, this should be "fairly trivial" to do (I assume he meant for someone far more smart and skilled than me) 

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Platform**
[ x ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

hendjoshsr71, 2021-09-29:
Link to the relevant ArduPilot Discuss post: [here](https://discuss.ardupilot.org/t/dji-hdl-protocol-in-ardupilot/76373)
Link to the relevant BetaFlight PR that added this feature: [here](https://github.com/betaflight/betaflight/pull/7881/files)

peterbarker, 2021-11-09:
@Luix333 it would be nice if you could test the referenced PR to see if it Works For You(tm).

PR #19159 (2021-11-18): AP_RCProtocol: add support for DJI Fast SBUS
It's just SBUS... but with increased opportunities for corruption

Fixes #18802
closed 2021-11-18 by peterbarker