ArduPilot/ardupilot #18920: Chibios build error

mundanePeo, 2021-10-13:
I type two command to build CUAVv5Nano
./waf configure --board CUAVv5Nano
./waf copter

but the log report a bug:
-----------------------------------------------------
Waf: Entering directory `/home/mjs/ardupilot/build/CUAVv5Nano'
Checking for env.py
env added HAL_NUM_CAN_IFACES=2
env added BOARD_FLASH_SIZE=2048
env added IOMCU_FW=0
env added APJ_BOARD_TYPE=STM32F767xx
env added PROCESS_STACK=0x1C00
env added HAS_EXTERNAL_FLASH_SECTIONS=0
env added USBID=0x1209/0x5740
env added MAIN_STACK=0x600
env added APJ_BOARD_ID=50
env added BOOTLOADER_EMBED=1
env added CORTEX=cortex-m7
env added FLASH_TOTAL=2064384
env added FLASH_RESERVE_START_KB=32
env added CHIBIOS_BUILD_FLAGS=USE_FATFS=yes MCU=cortex-m7 ENV_UDEFS=-DCHPRINTF_USE_FLOAT=1 CHIBIOS_PLATFORM_MK=os/hal/ports/STM32/STM32F7xx/platform.mk CHIBIOS_STARTUP_MK=os/common/startup/ARMCMx/compilers/GCC/mk/startup_stm32f7xx.mk
env added EXTERNAL_PROG_FLASH_MB=0
env appended CPU_FLAGS=['-mcpu=cortex-m7', '-mfpu=fpv5-d16', '-mfloat-abi=hard', '-DARM_MATH_CM7', '-u_printf_float']
env added OPTIMIZE=-O2
env added PERIPH_FW=0
env added WITH_FATFS=1
Padded 24 bytes for bootloader.bin to 16480
Embedding file bootloader.bin:/home/mjs/ardupilot/Tools/bootloaders/CUAVv5Nano_bl.bin
Embedding file hwdef.dat:/home/mjs/ardupilot/build/CUAVv5Nano/hw.dat
Embedding file io_firmware.bin:Tools/IO_Firmware/iofirmware_lowpolh.bin
[ 3/10] Creating build/CUAVv5Nano/hwdef.h
[ 4/10] Creating build/CUAVv5Nano/modules/ChibiOS/include_dirs
Including /home/mjs/ardupilot/libraries/AP_HAL_ChibiOS/hwdef/fmuv5/hwdef.dat
Adding environment OPTIMIZE -O2
Removing PE3
Removing IOMCU_UART
Removing UART8_TX
Removing UART8_RX
Removing AP_FEATURE_SBUS_OUT
Removing PA5
Removing PB3
Removing PB11
Removing FMU_CAP1
Removing FMU_CAP2
Removing FMU_CAP3
Removing PG9
Removing USART6_RX
Removing HAL_BATT_VOLT_SCALE
Removing HAL_BATT_CURR_SCALE
Removing HAL_GPIO_A_LED_PIN
Removing HAL_GPIO_B_LED_PIN
Setup for MCU STM32F767xx
Writing hwdef setup in /home/mjs/ardupilot/build/CUAVv5Nano/hwdef.h
MCU Flags: cortex-m7 ['-mcpu=cortex-m7', '-mfpu=fpv5-d16', '-mfloat-abi=hard']
Writing DMA map
Generating ldscript.ld
No default parameter file found

[ 5/10] Compiling libraries/AP_Scripting/generator/src/main.c
[ 6/10] Processing modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
Validation skipped for /home/mjs/ardupilot/modules/mavlink/message_definitions/v1.0/ardupilotmega.xml.
Parsing /home/mjs/ardupilot/modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
Validation skipped for /home/mjs/ardupilot/modules/mavlink/message_definitions/v1.0/common.xml.
Parsing /home/mjs/ardupilot/modules/mavlink/message_definitions/v1.0/common.xml
Validation skipped for /home/mjs/ardupilot/modules/mavlink/message_definitions/v1.0/uAvionix.xml.
Parsing /home/mjs/ardupilot/modules/mavlink/message_definitions/v1.0/uAvionix.xml
Validation skipped for /home/mjs/ardupilot/modules/mavlink/message_definitions/v1.0/icarous.xml.
Parsing /home/mjs/ardupilot/modules/mavlink/message_definitions/v1.0/icarous.xml
Validation skipped for /home/mjs/ardupilot/modules/mavlink/message_definitions/v1.0/minimal.xml.
[... 59 lines cut ...]
  876 |   case PWM_OUTPUT_ACTIVE_HIGH:
      |   ^~~~
../../modules/ChibiOS/os/hal/ports/STM32/LLD/TIMv1/hal_pwm_lld.c:884:10: error: this statement may fall through [-Werror=implicit-fallthrough=]
  884 |     ccer |= STM32_TIM_CCER_CC3P;
      |     ~~~~~^~~~~~~~~~~~
../../modules/ChibiOS/os/hal/ports/STM32/LLD/TIMv1/hal_pwm_lld.c:886:3: note: here
  886 |   case PWM_OUTPUT_ACTIVE_HIGH:
      |   ^~~~
../../modules/ChibiOS/os/hal/ports/STM32/LLD/TIMv1/hal_pwm_lld.c:894:10: error: this statement may fall through [-Werror=implicit-fallthrough=]
  894 |     ccer |= STM32_TIM_CCER_CC4P;
      |     ~~~~~^~~~~~~~~~~~~
../../modules/ChibiOS/os/hal/ports/STM32/LLD/TIMv1/hal_pwm_lld.c:896:3: note: here
  896 |   case PWM_OUTPUT_ACTIVE_HIGH:
      |   ^~~~
cc1: all warnings being treated as errors
make: *** [modules/ChibiOS/obj/hal_pwm_lld.o] Error 1
make: *** 正在等待未完成的任务....

Waf: Leaving directory `/home/mjs/ardupilot/build/CUAVv5Nano'
Build failed
 -> task in 'ChibiOS_lib' failed (exit status 2): 
	{task 140617571305184: ChibiOS_lib hwdef.h,hw.dat,include_dirs,board.c,board.h,bouncebuffer.c,bouncebuffer.h,chconf.h,ffconf.h,flash.c,flash.h,halconf.h,hrt.c,hrt.h,malloc.c,mcuconf.h,ppm.h,spi_hook.h,stdio.h,stm32_util.c,stm32_util.h,stm32f1_mcuconf.h,stm32f3_mcuconf.h,stm32f47_mcuconf.h,stm32g4_mcuconf.h,stm32h7_mcuconf.h,stm32l4_mcuconf.h,stubs.c,usbcfg.c,usbcfg.h,usbcfg_common.c,usbcfg_dualcdc.c,watchdog.c,watchdog.h,chibios_board.mk,chibios_common.mk,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,hwdef.h,stm32h7_mcuconf.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,buzzer.c,buzzer.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,board.c,board.h,hal.h,hal_adc.h,hal_buffers.h,hal_can.h,hal_channels.h,hal_crypto.h,hal_dac.h,hal_efl.h,hal_eicu.h,hal_files.h,hal_flash.h,hal_gpt.h,hal_i2c.h,hal_i2s.h,hal_icu.h,hal_ioblock.h,hal_mac.h,hal_mii.h,hal_mmc_spi.h,hal_mmcsd.h,hal_objects.h,hal_pal.h,hal_persistent.h,hal_pwm.h,hal_queues.h,hal_rtc.h,hal_sdc.h,hal_serial.h,hal_serial_usb.h,hal_sio.h,hal_spi.h,hal_st.h,hal_streams.h,hal_trng.h,hal_uart.h,hal_usb.h,hal_usb_cdc.h,hal_wdg.h,hal_wspi.h,hal_mfs.c,hal_mfs.h,hal_flash_device.c,hal_flash_device.h,hal_flash_device.c,hal_flash_device.h,hal_serial_nor.c,hal_serial_nor.h,hal_i2c_lld.c,hal_i2c_lld.h,chprintf.c,chprintf.h,memstreams.c,memstreams.h,nullstreams.c,nullstreams.h,osal_vt.c,osal_vt.h,osal.c,osal.h,osal.c,osal.h,osal.c,osal.h,aducm_cc.h,aducm_gp.h,aducm_isr.c,aducm_isr.h,hal_lld.c,hal_lld.h,hal_pal_lld.c,hal_pal_lld.h,hal_serial_lld.c,hal_serial_lld.h,hal_spi_lld.c,hal_spi_lld.h,hal_st_lld.c,hal_st_lld.h,aducm_gp.h,aducm_isr.c,aducm_isr.h,hal_lld.c,hal_lld.h,hal_st_lld.c,hal_st_lld.h,avr_pins.h,avr_timers.h,hal_lld.c,hal_lld.h,hal_adc_lld.c,hal_adc_lld.h,hal_ext_lld.c,hal_ext_lld.h,hal_pal_lld.c,hal_pal_lld.h,hal_i2c_lld.c,hal_i2c_lld.h,hal_spi_lld.c,hal_spi_lld.h,hal_gpt_lld.c,hal_gpt_lld.h,hal_icu_lld.c,hal_icu_lld.h,hal_pwm_lld.c,hal_pwm_lld.h,hal_st_lld.c,hal_st_lld.h,hal_serial_lld.c,hal_serial_lld.h,hal_usb_lld.c,hal_usb_lld.h,hal_lld.c,hal_lld.h,hal_pal_lld.c,hal_pal_lld.h,hal_st_lld.c,hal_st_lld.h,hal_uart_lld.c,hal_uart_lld.h,avr_pins.h,avr_timers.h,hal_lld.c,hal_lld.h,hal_crypto_lld.c,hal_crypto_lld.h,hal_dac_lld.c,hal_dac_lld.h,xmega_dma_lld.c,xmega_dma_lld.h,hal_pal_lld.c,hal_pal_lld.h,hal_spi_lld.c,hal_spi_lld.h,hal_st_lld.c,hal_st_lld.h,hal_serial_lld.c,hal_serial_lld.h,hal_uart_lld.c,hal_uart_lld.h,hal_wdg_lld.c,hal_wdg_lld.h,hal_lld.c,hal_lld.h,hal_pal_lld.c,hal_pal_lld.h,hal_serial_lld.c,hal_serial_lld.h,hal_spi_lld.c,hal_spi_lld.h,hal_st_lld.c,hal_st_lld.h,vic.c,vic.h,hal_spi_lld.c,hal_spi_lld.h,spc5_dspi.h,spc5_edma.c [... cut ...] m.mk,platform.mk,platform_f105_f107.mk,platform.mk,platform.mk,platform.mk,platform.mk,platform.mk,platform.mk,platform.mk,platform.mk,platform.mk,platform.mk,platform_l432.mk,platform.mk,platform.mk,platform.mk,platform.mk,board.mk,osal.mk,platform.mk,ap_romfs_embedded.h -> libch.a}
 (run with -v to display more information)
----------------------------------------------------------------------------------------------------------------------------
Could you tell me how to solve the problem? tks

khancyr, 2021-10-13:
hello,

I cannot replicate even with python2. Try to clean your build with 
`./waf distclean`
and update the submodules
`git submodule update --init --recursive`
it should be working.

otherwise check that `/usr/lib/ccache/arm-none-eabi-gcc -v`   is 10.2.1 too

Robin-Chenxb, 2023-05-08:
Hello， 
I have the same problem as you.
Have you solved this problem yet? How did it solve?

andyp1per, 2023-05-08:
minimum required version is now python 3.8. You are going to struggle on 18.04 -better to use a later version of ubuntu