ArduPilot/ardupilot #18921: Please add BeagleBoneBlue binaries as a build target or update the docu

drtrigon, 2021-10-13:
Trying to setup a BeagleBoneBlue with ArduPilot is not possible due to old instructions and missing links.

If you look at https://ardupilot.org/copter/docs/common-beagle-bone-blue.html you will find instruction leading you to the information on github https://github.com/mirkix/ardupilotblue. At some point you are instructed to download the ArduPilot prebuilt binary from http://bbbmini.org/download/blue/. The issue with this is that this link is dead.

I can think of 2 possible solutions:
1. make the link work again or may be even better setup the automatic building procedure such that the needed BeagleBoneBlue binaries are generated and put into https://firmware.ardupilot.org/ like all other targets
2. update the documentation such that it contains detailed information and steps on how to build these binaries

As a third alternative it could be included into the custom firmware builder https://custom.ardupilot.org/

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

drtrigon, 2021-10-14:
Meanwhile I found 2 possible ressources, but they do not fully agree on the command sequence to run:

- https://github.com/mirkix/BBBMINI/blob/master/doc/software/software.md#compile-ardupilot-native-on-beaglebone
- https://ardupilot.org/dev/docs/building-for-beaglebone-black-on-linux.html#installing-and-making-ardupilot-on-bbb (here "bbb" = Beagle Bone Black NOT Blue)

Can someone shed some light on this and explain what the correct sequence is nowadays?

drtrigon, 2021-11-30:
@juvinski: Thanks for your hints, we will test this as soon as possible and report back. The link to imfatants repo is very useful and what we worked with - but it also contains dead links and outdated info.

Thus I think the ArduPilot wiki really needs to be updated, e.g. with the info you posted here.

drtrigon, 2021-12-01:
Better that the link to imfatants repo is this one: https://sudonull.com/post/8513-ArduPilot-for-beginners-Installation-and-configuration-on-BeagleBone-Blue

drtrigon, 2021-12-23:
Our step by step manual: https://gitlab.com/fhgr/beagleboneblue-fc

drtrigon, 2021-12-23:
I would be willing to set up the automatic cross compilation to build binaries.

If someone explains me what system is used for the other build targets, where the sources/repos are etc. I can try to implement it in the existing build system.

Alternatively I can set it up using https://travis-ci.org.

drtrigon, 2021-12-23:
Ok, I've again learned that I am quite old and travis-ci does - of curse - not exist anymore. But on the other hand github introduced actions. ;)

I created this repo: https://github.com/drtrigon/ardupilot-beagle-bone-blue-binaries

The built target binaries can be found by going to: https://github.com/drtrigon/ardupilot-beagle-bone-blue-binaries/actions
There select the latest/topmost from the list and on the following page you can find the binaries under artifacts at the bottom of the summary page.

The binaries do get built twice a day. `latest` is `master` and contains all binaries. The other ones contain the given release and vehicle type only.

Have fun and please give feedback. Very nice would be if we could include that into the existing ardupilot build host.

drtrigon, 2022-01-07:
Pull request moved: https://github.com/ArduPilot/ardupilot/pull/19693

drtrigon, 2022-01-07:
@khancyr: What would be needed to add it to https://custom.ardupilot.org/ ? This is of minor priority but should finally also be done.

drtrigon, 2022-01-09:
The PR has been merged. Thanks a lot to everybody involved!

drtrigon, 2022-01-11:
@khancyr: Sorry for re-opening, but I wonder whether it's possible to trigger a build on the firmware server such that we get access to the 4.1.6 binary for the blue target? (While I do build them myself, I prefer to get it from firmware server in order to have an official reference and make sure everybody gets the exactly same state.)

khancyr, 2022-01-11:
@drtrigon the firmware server currently go some issue, we are looking at solving them

drtrigon, 2022-01-11:
@khancyr: I'm sorry to hear that - thank for your effort! Looking forward to see this working again...

drtrigon, 2022-01-12:
Works again! Thanks a lot!

closed 2022-01-12 by drtrigon