ArduPilot/ardupilot #18932: [VTOL] LOITER_TIME after a DO_TRANSITION_VTOL at 3 (MC) crashes the code

robustini, 2021-10-14:
During an automatic mission, which I attach, if I put the tailsitter in LOITER_TIME after a DO_TRANSITION_VTOL at 3 (MC) the code crashes, or at least this happens with SITL and Realight.
To check if this bug really exists in the code, or if maybe it's just a communication problem with flightaxis.
@tridge used this thing with quadplane and found no problems.
Maybe it's just a problem with the SITL, but better check because here it can always be replicated.
Tested at [this commit](https://github.com/ArduPilot/ardupilot/commit/f076bb3ad1eac2621f0aa7faa5ad5ff819de76d3).

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
CAT Tailsitter

**Hardware type**
SITL and Realflight 9

**Logs**
[00000003.ZIP](https://github.com/ArduPilot/ardupilot/files/7345836/00000003.ZIP)
[tailsitter_mission.zip](https://github.com/ArduPilot/ardupilot/files/7345839/tailsitter_mission.zip)

peterbarker, 2021-10-15:
Thanks @robustini - I replicated it locally here.

I was unable to replicate on master - whatever parameters come in `SITL_Models` are insufficient to make the SkyCat fly on master.  I replicated it on Plane-4.1.1 instead.

```
PANIC: internal error 1048576
Running: sh Tools/scripts/dumpstack.sh 25597 >dumpstack_arduplane.25597.out 2>&1
Stack dumped
Could not attach to process.  If your uid matches the uid of the target
process, check the setting of /proc/sys/kernel/yama/ptrace_scope, or try
again as the root user.  For more details, see /etc/sysctl.d/10-ptrace.conf
warning: process 25597 is already traced by process 25583
ptrace: Operation not permitted.
/tmp/gdb.27530:2: Error in sourced command file:
No stack.
^C
Thread 1 "arduplane" received signal SIGINT, Interrupt.
AP_HAL::panic (errormsg=0x5555558d0d95 "internal error %u")
    at ../../libraries/AP_HAL_SITL/system.cpp:46
46          for(;;);
(gdb) bt
#0  AP_HAL::panic (errormsg=0x5555558d0d95 "internal error %u")
    at ../../libraries/AP_HAL_SITL/system.cpp:46
#1  0x00005555556902c1 in AP_InternalError::error (
    this=0x555555bc9e18 <instance>, 
    e=AP_InternalError::error_t::flow_of_control, line=2643)
    at ../../libraries/AP_InternalError/AP_InternalError.cpp:20
#2  0x00005555555c76c5 in QuadPlane::vtol_position_controller (
    this=0x555555bc91c8 <plane+31592>) at ../../ArduPlane/quadplane.cpp:2643
#3  0x00005555555c9494 in QuadPlane::control_auto (
    this=0x555555bc91c8 <plane+31592>) at ../../ArduPlane/quadplane.cpp:3208
#4  0x00005555555bb007 in ModeAuto::update (this=0x555555bc79f0 <plane+25488>)
    at ../../ArduPlane/mode_auto.cpp:57
#5  0x00005555555a1942 in Plane::update_control_mode (
    this=0x555555bc1660 <plane>) at ../../ArduPlane/ArduPlane.cpp:444
#6  0x00005555555a2675 in Functor<void>::method_wrapper<Plane, &Plane::update_control_mode> (obj=0x555555bc1660 <plane>)
    at ../../libraries/AP_HAL/utility/functor.h:88
#7  0x00005555556477a8 in Functor<void>::operator() (
    this=0x555555baa760 <Plane::scheduler_tasks+128>)
    at ../../libraries/AP_HAL/utility/functor.h:54
#8  0x0000555555689a2f in AP_Scheduler::run (this=0x555555bc1928 <plane+712>, 
    time_available=3333) at ../../libraries/AP_Scheduler/AP_Scheduler.cpp:198
#9  0x0000555555689ec6 in AP_Scheduler::loop (this=0x555555bc1928 <plane+712>)
    at ../../libraries/AP_Scheduler/AP_Scheduler.cpp:323
#10 0x000055555568f4f3 in AP_Vehicle::loop (this=0x555555bc1660 <plane>)
    at ../../libraries/AP_Vehicle/AP_Vehicle.cpp:182
#11 0x0000555555786826 in HAL_SITL::run (
    this=0x555555bd9240 <AP_HAL::get_HAL()::hal>, argc=11, 
    argv=0x7fffffffd988, callbacks=0x555555bc1660 <plane>)
    at ../../libraries/AP_HAL_SITL/HAL_SITL_Class.cpp:251
#12 0x00005555555a25c2 in main (argc=11, argv=0x7fffffffd988)
    at ../../ArduPlane/ArduPlane.cpp:694
(gdb) 
```

It looks like the code in question knows something is wrong, tries to fix it and then raises the internal error.  Taking the internal error out allows the vehicle to complete a safe-ish looking landing, so this isn't a trivial fall-out-of-sky bug on a real vehicle.

```
    case QPOS_NONE:
        poscontrol.set_state(QPOS_POSITION1);
        INTERNAL_ERROR(AP_InternalError::error_t::flow_of_control);
        break;
```

@IamPete1 @tridge

IamPete1, 2021-10-15:
Since its a QPOS error I would be surprised if it was tailsitter specific, I would guess normal Q planes too. I would also guess any of the loiter commands would do the same, turns, alt and unlimited. 

I will take a closer look in the morning.

IamPete1, 2021-10-16:
You don't even need the loiter, VTOL takeoff followed by VTOL land will also cause this.