ArduPilot/ardupilot #18952: set_target_altitude_current method in guided mode

ZeinabJalali4, 2021-10-16:
## Bug report

**Issue details**

We have a function that uses guided mode for controlling a plane. It updates the goal location alternately, but it can't reach the goal alt. 
I started reading the source of Guided mode but I couldn't understand why there is a set_target_altitude_current method in function set_guided_WP in Guided mode. According to comments, it uses this function for modes like Cruise. But I can't understand the usage of this in Guided mode.
I commented this line in Guided and my problem was solved and the plane reaches the alt as requested. I don't know whether it is true to comment this line or not? Would then be any problem?

**Version**
ArduPlane 4.0.9

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ × ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
SITL

magicrub, 2021-10-18:
Please post a link to the specific line you're talking about

ZeinabJalali4, 2021-11-16:
Does anyone have any idea about this question?
@rmackay9 @magicrub