ArduPilot/ardupilot #19103: Leak failsafe from ardusub for Arduboat

geofrancis, 2021-11-01:
can the leak failsafe from ardusub be added to the regular rover firmware so it can be used on boats?

rmackay9, 2021-11-02:
@geofrancis,

It's certainly possible and during my boat mapping last week I was thinking it would be very reassuring to have at least reporting of leaks (using the AP_LeakDetector feature).

I strongly suspect that we will add this eventually it just depends upon finding a dev to take on the effort.

geofrancis, 2021-11-02:
the 2 ideas i was looking at was using a esp32 companion that will be on the boat to manage the bilge and alerts for it or use one of these little rain detector boards use it to trigger a gpio button on the pixhawk set to something like RTL along with triggering a relay or mosfet for a pump, that way if it clears the water it should switch back out of rtl and into whatever mode it was in. but if it can't it will continue RTL. its a hack but it should work until AP_LeakDetector is implimented.

rmackay9, 2021-11-02:
@geofrancis, Lua scripts might also be able to do some of what you're suggesting without the need for a companion esp32.. but whatever works ..

geofrancis, 2021-11-02:
@rmackay9 the esp32 will be on the boat anyway for managing a bunch of water and air sensors and any other stuff that ends up on it so it wouldnt have taken much to add another sensor but it would still need to interface with ardupilot to failsafe.

geofrancis, 2022-01-18:
@rmackay9 do you think this is best done in lua and I should close this or is it still something you would want to integrate into the firmware?

ES-Alexander, 2024-06-27:
Agreed that this is a valuable feature, possibly even for ground-based rovers in case they get caught in heavy rain or something.

There's some additional discussion in [this forum thread](https://discuss.bluerobotics.com/t/leak-detection-with-ardurover-on-blueboat/16540) about a current Lua script approach for the BlueBoat, although the implementation there is just an alert at this stage, and only works for the Navigator flight controller board.

> do you think this is best done in lua and I should close this or is it still something you would want to integrate into the firmware?

The `AP_LeakDetector` library [has been extended](https://github.com/ArduPilot/ardupilot/pull/22014) since this Issue was first raised, and it seems worth making use of rather than replicating in a Lua script.

ES-Alexander, 2024-06-27:
I had a brief look at the code, and it doesn't look like it'll be overly difficult to include (just a bunch of small things in several files), but the Rover codebase is ahead of the Sub codebase by enough that there are some structure and feature changes that should be factored in as part of a port, so it's not quite a copy-paste operation.

Unfortunately I don't feel up to implementing it at the moment, but (assuming nobody hops in to do it in the meantime) I'll see if I can get this added to the Blue Robotics software goals within the next few quarters, since this seems like a feature we should care about for BlueBoat, and given we sell leak sensors/probes there's some extra motivation.