ArduPilot/ardupilot #19104: Depth Fence for Arduboat

geofrancis, 2021-11-01:
Now that rover firmware can read water depth it really needs a way of avoiding shallow water and running aground, the simplest way of doing this would be an altitude fence so its triggered if sonar depth is below a set value.  Currently, all altitude fences have been disabled on Rover firmware so there is no way of setting any kind of vertical limits on rovers.

IamPete1, 2021-11-01:
The problem is deciding what to do, because the reading is for depth under the boat, there is no inherent direction. You cannot just move away from the obstacle like you might with a forward facing rangefinder. You could go back the way you came in, but that does not allow to move past the obstacle. 

If we were to support some a system that can read the depth in-front of the vehicle, we can treat it the same as rangefinder based avoidance, and it should all work already.

geofrancis, 2021-11-01:
A couple of options I can think of would be:

1. just stop at the line, essentially just loiter when it gets too shallow
2. RTL/rallly point if  it gets too shallow
3. Smart RTL to find a safe way back.
4, tack if its sailing

rmackay9, 2021-11-01:
Thanks for raising this issue.  I agree with the comments above that there are two possible solutions and theoretically either or both could work.  A fence or failsafe could trigger a user configurable action or we could use a horizontally facing underwater sonar and then do avoidance.

geofrancis, 2021-11-02:
the problem i see with the forward looking sonar is that it wont really work for gentle slopes 

> 
> 
> Thanks for raising this issue. I agree with the comments above that there are two possible solutions and theoretically either or both could work. A fence or failsafe could trigger a user configurable action or we could use a horizontally facing underwater sonar and then do avoidance.

The problem I see with the forward-looking sonar is that they generally have a very wide beam, I think it might start picking up surface reflections and objects in front rather than below as it's tilted up more, but I'm not a sonar expert. The other issue is it will interfere with the operation of the above surface ultrasonic avoidance, as I don't think I would be able to set different distances for above and below forward-looking range finders.

rmackay9, 2021-11-02:
@geofrancis,

Yes, and our AP_Proimity library only supports a single sensor at the moment so we couldn't do below surface and above surface detection and avoidance separately (yet).

By the way, for above surface avoidance, I think we should use radar (which may not be confused by the surface of the water) and/or the OAK-D (AI, 3D camera) which will hopefully we able to recognise objects and then provide a distance for them.

geofrancis, 2021-11-02:
> By the way, for above surface avoidance, I think we should use radar (which may not be confused by the surface of the water) and/or the OAK-D (AI, 3D camera) which will hopefully we able to recognise objects and then provide a distance for them.

3d AI is far outside my price and power budget unfortunately, but the radar sounds very interesting, I was not aware of any that were currently compatible with ardupilot apart from some altimeters. With all the new cars getting radars I am surprised some automotive radars haven't filtered down to hobbyists yet.

geofrancis, 2021-12-15:
has anyone ever tested one of the waterproof lidars like the TF-mini plus under the water? im aware that the water will diffuse the light much faster but how much faster?

paularthur, 2022-01-17:
Thanks people. We're doing mapping and water quality testing on a shallow muddy harbour - we're going to run aground...
This would help a lot - and agreed - the action to take is not simple - but run a script would allow us to move forward.
For us - tide is important  - incoming (just wait a bit) or outgoing (wait 6hrs).
or even
If _too_shallow_ then _skip to the next waypoint_ - kind of works - but also fails horribly.

geofrancis, 2022-01-18:
I have added a lua script to test at https://discuss.ardupilot.org/t/shallow-depth-warning-and-avoidance-using-lua/78545/8 I think that is going to be the best way of handling this issue rather than something integrated into ardupilot is its a very specialized task.

I'm closing this issue and will continue discussion at the link above.

closed 2022-01-18 by geofrancis