ArduPilot/ardupilot #19110: APM/LOGS directory cannot be created on Pixhawk1

Jaaaky, 2021-11-02:
## Bug report
**Issue details**

Building ArduPlane master for Pixhawk1, all paramaters are default, with freshly formatted SD card I get
`GCS shows "Failed to create log directory /APM/LOGS : EIO"` on mavproxy
-  I can confirm LOG_BACKEND_TYPE=1 .. it's the default
- Parameters are all default, no code or parameter changed at all for Pixhawk1 board .. when I simply run arm throttle on mavproxy I get:
```
Failed to create log directory /APM/LOGS : EIO
Log open fail for /APM/LOGS/00000001.BIN - ENOENT
Log open fail for /APM/LOGS/00000001.BIN - ENOENT
```

And the message:
`Failed to create log directory /APM/LOGS : EIO` Keeps showing every few seconds ..

- It created APM/STRG_BAK normally, so issue is LOGS specific
- Creating the LOGS directory on computer allows the log to work fine
- The following patch solves the problem for me:

```
diff --git a/libraries/AP_Logger/AP_Logger_File.cpp b/libraries/AP_Logger/AP_Logger_File.cpp
index d39262dec4..9589d1c273 100644
--- a/libraries/AP_Logger/AP_Logger_File.cpp
+++ b/libraries/AP_Logger/AP_Logger_File.cpp
@@ -45,6 +45,7 @@ AP_Logger_File::AP_Logger_File(AP_Logger &front,
     AP_Logger_Backend(front, writer),
     _log_directory(log_directory)
 {
+    ensure_log_directory_exists();
     df_stats_clear();
 }
 
```

**Version**
master commit number dfd7dfc1af4dd74225cf697103b658d355e09fd3

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**
Pixhawk1

IamPete1, 2021-11-14:
https://github.com/ArduPilot/ardupilot/issues/18737
#17257
#16834
#16103
#14860

tridge, 2021-11-15:
I have reproduced this

tridge, 2021-11-15:
fixed by #19267

PR #19267 (2021-11-16): ChibiOS: fixed f_mkdir DMA allocation
this ensures we have DMA capable memory for f_mkdir

fixes #19110
commit bba0384e (2021-11-16): ChibiOS: fixed f_mkdir DMA allocation

this ensures we have DMA capable memory for f_mkdir

fixes #19110
closed 2021-11-15 by tridge