ArduPilot/ardupilot #19119: SIM_RC_FAIL = 2 does not trigger RC failsafe

shiv-tyagi, 2021-11-02:
## Bug report

**Issue details**

`SIM_RC_FAIL` parameter when set to 2 (i.e., SITL_RCFail_Throttle950) does not trigger throttle failsafe. 

Steps to reproduce:
- param set FS_THR_VALUE 960
- param set FS_THR_ENABLE 1
- param set SIM_RC_FAIL 2
- arm throttle (to confirm if arming checks fail because of below acceptable throttle)
[image]

**Version**
master

**Platform**
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

rmackay9, 2021-11-03:
I did a quick test with master Copter and it works OK.
[image]

closed 2021-11-03 by rmackay9

shiv-tyagi, 2021-11-03:
@rmackay9 I tested it again and found one interesting thing. It does not work if you set guided mode and try to takeoff with `takeoff` command but it works when we takeoff by manually setting throttle channel to some value in a mode like loiter (maybe this is how you tested it).
I am not sure if we need to worry about this behaviour.

rmackay9, 2021-11-03:
@shiv-tyagi,

This is probably the expected behaviour but maybe have a peek at the FS_OPTIONS parameter.