ArduPilot/ardupilot #19124: Potential Barometer Driver issue with iFlight Beast F7 due to board redesign

acloitre, 2021-11-02:
Hello,

I just flashed Ardupilot on my iFlight Beast F7 AIO 45A v2 but the board outputs the same error, regardless of the firmware I use (stable, beta or master):

> Config error: Baro: unable to initialise driver

I believe it is due to a hardware redesign from iFlight, as they state the following on the product page: https://shop.iflight-rc.com/beast-f7-45a-v1-1-aio-mpu6000-25-5x25-5-pro1377 (tab 'specification', paragraph 'Change log')

> V1.0 - V1.1: Due to some global supply issues, the Barometer and Flash component was updated as well as some circuit improvements implemented.
> 
> Before BMP280 changed to DPS310
> Before 8MB Flash changed to 16MB Flash

I checked the hwdef file for this firmware: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_ChibiOS/hwdef/BeastF7/hwdef.dat
And line 129 reads the following:

> BARO BMP280 I2C:0:0x76

Could a new firmware be released, that would support the v2 version of the board, that has a DPS310 barometer instead of a BMP280?

Best regards,

Audren

acloitre, 2021-11-02:
I forked ardupilot and compiled it for the BeastF7 after changing line 129 with the following:

> BARO DPS310 I2C:0:0x76

The barometer error disappeared but I now have the following new error:

> Config error: INS: unable to initialise driver

Looking around online, many people recommend to power the board from battery prior to connecting via USB. I tried, but this did not fix the problem.
Is there another setting you can think of that would need to be updated for the BeastF7 firmware to work with the version v2 of the board?

andyp1per, 2021-11-03:
As far as I can tell the IMU is the same so I am not sure why you might get that error. We can make the barometer change no problem. Betaflight has a new hw definition which includes a second gyro - I wonder if that is what is needed.

Try adding this:
```
PB12 MPU6000_2_CS CS
PB13 SPI2_SCK SPI2
PB14 SPI2_MISO SPI2
PB15 SPI2_MOSI SPI2

SPIDEV mpu6000_2   SPI2 DEVID3 MPU6000_2_CS MODE3  1*MHZ  4*MHZ
IMU Invensense SPI:mpu6000_2 ROTATION_YAW_270
```

acloitre, 2021-11-03:
Thank you for the fast reply Andy.
I implemented the change you suggested but I still see the same issue. 

I opened up my quad to double-check the flight controller, and iFlight installed the newer 55A-ready version of their flight controller instead of the 45A version that is marketed on the box.

- [45A version product page](https://shop.iflight-rc.com/beast-f7-45a-v1-1-aio-mpu6000-25-5x25-5-pro1377) and [associated hwdef file from BetaFlight](https://github.com/betaflight/unified-targets/blob/master/configs/default/IFRC-IFLIGHT_F745_AIO.config)
- [55A version product page](https://shop.iflight-rc.com/quad-parts-cat20/electronics-cat27/beast-series-cat330/Beast-F7-45A-V2-AIO-MPU6000-25.5-25.5-Pro1625) and [associated hwdef file from BetaFlight](https://github.com/betaflight/unified-targets/blob/master/configs/default/IFRC-IFLIGHT_F745_AIO_V2.config)

On top of changing the barometer, iFlight also changed the IMU on the 55A version. The new IMU is a Bosch BMI270, different from the expected MPU6000 defined in the hwdef file for the BeastF7 firmware.

Is this new IMU compatible with Ardupilot? If so, what would be the chagne in the hwdef file for Ardupilot?

Thank you for all your help with this.

andyp1per, 2021-11-03:
Unfortunately we don't currently have a BMI270 driver and it looks like the BMI160 driver will not work.

acloitre, 2021-11-03:
OK, thank you very much Andy. 
I will look for a 45A version of the Beast F7 to put as a replacement part in my drone, and I will keep an eye out to see if Ardupilot adds support for the BMI270 in the future.
I appreciate all the help you provided, thanks again!

andyp1per, 2021-11-03:
I looked at the betaflight code - the BMI270 driver is very different. I suspect it might be a while unless some partners demand support for this. Sorry.

acloitre, 2021-11-03:
No worries, I am glad we could find the source of the issue quickly!

acloitre, 2021-11-05:
Hi Andy,

I am reopening the issue because I am still getting some troubles setting up the BeastF7, even after getting an older board that has an MPU6000. I built a custom firmware using the fix for the DPS310 barometer and the board boots up without errors from the barometer anymore.
I however cannot setup the board in QGC as it says:
`QGroundControl does not currently support setup of your vehicle type. If your vehicle is already configured you can still fly.`

I do get two errors when flashing my custom firmware:

> Firmware file missing parameter_xml_size key
> Firmware file missing airframe_xml_size key

And the log shows that the firmware or FRAME_CLASS should be checked:
[image]

The FRAME_CLASS is set to 0 currently, and trying to set it to 1 (Quad) in the parameters tab does not seem to work.

Is there a setting I am missing when building the firmware? I used the following commands to build it:
```
./waf configure --board BeastF7
./waf copter
```
I will appreciate any help you can share.

andyp1per, 2021-11-06:
Try using mavproxy to set the frame class - there is an issue with MP and QGC that they cannot read/wriet params until you set the frame class

acloitre, 2021-11-08:
Thank you very much Andy.
Manually setting the FRAME_CLASS parameter via MAVProxy solved the problem. I can now access the board via QGC and proceed with the board setup.

closed 2021-11-08 by acloitre