ArduPilot/ardupilot #19147: AirSim Thrust Loss Error Prevents Arming

khnunn, 2021-11-04:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**
Hi, I am getting a rather strange error when running AirSim. When I arm throttle, the MavProxy states "AP: Potential Thrust Loss (6)". The rotors do not spin up in the simulation. This started after performing radio calibrations for a Taranis X9D+ 2019 in Mission Planner.

I have looked at my parameter settings, FLIGHT_OPTIONS is set to 0 which should disable this warning. I've attached my parameter settings (as a .txt file), which show that RC trim values are indeed at the midpoint for each PWM channel  (rather than below the RC minimum). Also prearm checks are disabled.

I have reset parameters fully, and tried a fresh re-install of ArduCopter 4.10. Nothing seems to work.

**Version**
ArduCopter v4.1.0

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
AirSim Hexacopter Model

**Hardware type**
PixHawk 2.1

**Logs**
[parameters.txt](https://github.com/ArduPilot/ardupilot/files/7477146/parameters.txt)

MavProxy console log:
```
Got COMMAND_ACK: DO_SET_MODE: ACCEPTED
Mode ALT_HOLD
Got COMMAND_ACK: COMPONENT_ARM_DISARM: ACCEPTED
AP: Arming motors
ARMED
Arming checks disabled
AP: Potential Thrust Loss (6)
```

IamPete1, 2021-11-04:
This is probably a motor ordering issues. Cant say what it should be for without knowing the airsim model. 

This is what AP is expecting:
https://ardupilot.org/copter/docs/connect-escs-and-motors.html#hexacopter

You can change round your servo output function to match that to the airsim model.

khnunn, 2021-11-04:
> 
> 
> This is probably a motor ordering issues. Cant say what it should be for without knowing the airsim model.
> 
> This is what AP is expecting: https://ardupilot.org/copter/docs/connect-escs-and-motors.html#hexacopter
> 
> You can change round your servo output function to match that to the airsim model.

I will try changing servo outputs. For my AirSim model I am using Chris Lovett's model from [here](https://github.com/microsoft/AirSim/wiki/hexacopter), and my settings.json is:

```
{
  "SettingsVersion": 1.2,
  "LogMessagesVisible": true,
  "SimMode": "Multirotor",
  "OriginGeopoint": {
    "Latitude": -35.363261,
    "Longitude": 149.165230,
    "Altitude": 583
  },
  "PawnPaths": {
    "DefaultQuadrotor": {"PawnBP": "Class'/AirSim/Blueprints/BP_FlyingHex.BP_FlyingHex_C'"}
  },
  "Vehicles": {
    "Copter": {
      "VehicleType": "ArduCopter",
      "RC": {
              "RemoteControlID": 0
              },
      "Pawnpath": "DefaultQuadrotor",
      "EnableCollisionPassThrough": false,
      "EnableCollisions": true,
      "IsFpvVehicle": false,
      "SerialBaudRate": 115200,
      "SerialPort": "COM8",
      "UseSerial": true,
      "LocalHostIp": "127.0.0.1",
      "UdpIp": "127.0.0.1",
      "UdpPort": 9003,
      "ControlPort": 9002,
      "Model": "Hexacopter"
    }
  }
}
```

khnunn, 2021-11-07:
I have resolved the issue. AirSim was not rebuilding EEPROM whenever I loaded a new set of parameters from a file. Adding the -w command to "../Tools/autotest/sim_vehicle.py -v ArduCopter -f airsim-copter --console --map -w" reset EEPROM and allowed me to define new parameters.

closed 2021-11-07 by khnunn