ArduPilot/ardupilot #19158: Mavlink rate divided by 4

zhouzhiwen2000, 2021-11-05:
## Bug report

**Issue details**

The mavlink send rate is divided by 4. When SR0 is set to 40Hz, the actual send rate is 10Hz.

**Version**
 4.1-stable and 4.2-dev

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
 quad

**Hardware type**
pixhawk4

amilcarlucas, 2021-11-05:
This is not a bug. If your channel does not have the requested datarate capability the performance will degrade.
Test it with a USB direct connection, making sure that the requested datarate is bellow 115200 baud.

closed 2021-11-05 by amilcarlucas

zhouzhiwen2000, 2021-11-05:
> This is not a bug. If your channel does not have the requested datarate capability the performance will degrade. Test it with a USB direct connection, making sure that the total SR0_* requested datarate is bellow 115200 baud.

I can't seem to find the code related to this behaviour, can you show it to me? Thanks a lot!

amilcarlucas, 2021-11-05:
Take a look at ardupilot/modules/mavlink/pymavlink/tools/mavtelemetry_datarates.py for a total datarate calculator.

zhouzhiwen2000, 2021-11-06:
> Take a look at ardupilot/modules/mavlink/pymavlink/tools/mavtelemetry_datarates.py for a total datarate calculator.
The weird thing is I am using 921600 baud rate and the calculated datarate is nowhere near this.

peterbarker, 2021-11-15:
On Thu, 4 Nov 2021, zhouzhiwen2000 wrote:

> The mavlink send rate is divided by 4. When SR0 is set to 40Hz, the actual send rate is 10Hz.

Are you sure your GCS isn't resetting that parameter from underneath you?

Setting the parameters directly can be frustrating when any GCS which pops 
up sets all streams to some rate.

Also - you can ask for a *specific* message to be streamed faster, which 
might be more useful for you; see MAV_CMD_SET_MESSAGE_INTERVAL

zhouzhiwen2000, 2021-11-15:
> On Thu, 4 Nov 2021, zhouzhiwen2000 wrote:
>
>
> Are you sure your GCS isn't resetting that parameter from underneath you?
>
> Setting the parameters directly can be frustrating when any GCS which pops
> up [...]

I was using mavros. And I checked, it's not setting the parameters.

zhouzhiwen2000, 2021-11-16:
I reconnected the serial port. It seems fine now, not sure how it works though.