ArduPilot/ardupilot #19160: Rover/Boat: gyro calibration fails if autopilot is rebooted while on the water

rmackay9, 2021-11-05:
Rover/Boat's gyro calibration (run soon after startup) nearly always fails if the boat is already on the water.  The means the user needs to struggle a bit to disable some arming checks before they can arm the vehicle.

If the vehicle is a boat we should somehow relax the gyro calibration so that it passes.

geofrancis, 2021-11-05:
this should be made into an actual mode rather than just a page on the wiki 
https://ardupilot.org/copter/docs/boat-mode.html

IamPete1, 2021-11-06:
We have [`INS_GYR_CAL`](https://ardupilot.org/rover/docs/parameters.html#ins-gyr-cal-gyro-calibration-scheme) Setting to 0 AP will never run gyro calibration on start up. 

We could set the default to 0 for boats I guess.

geofrancis, 2021-11-15:
this has always baffled me, how do the dji machines get away without a gyro calibration?

rmackay9, 2021-11-15:
@geofrancis, I'm not sure they do.. at least when a photographer was filming my DeSET mapping boat we had to make a special trip to put the drone on shore to allow it to arm and takeoff.

geofrancis, 2021-11-16:
> 
> 
> @geofrancis, I'm not sure they do.. at least when a photographer was filming my DeSET mapping boat we had to make a special trip to put the drone on shore to allow it to arm and takeoff.
> 
> I think we should add a special case for boats that still does a gyro call but if it fails for too long just does a much more relaxed version.

I have flown phantoms from cruisers before, but given what you're saying, I suspect it was probably just a big enough boat. 

As far as I can think, the gyros aren't really used much currently other than yaw, and possibly roll when sailing.  I think there are some open requests for using pitch and roll stabilization on boats with trim tabs so the gyro might become more important later. if that gets implemented.

RK425, 2022-08-24:
Hi All, 
Is there anymore information on this? What is the impact on acro modes and auto modes if the gyros aren't calibrated on startup?  I also have a watercraft and if it goes into a restart the gyros fail and I have to tow it to shore before it will calibrate.

IamPete1, 2022-08-24:
I doubt it makes a noticeable defence, the EKF will learn the correct offsets quite quickly. We could add a "learn and save" option like the compass one we have, so that if you do have to restart for any reason you come back up with the correct value right away.