ArduPilot/ardupilot #19163: Ardupilot4.1.0 firmware flashes incorrectly (default params not set up correctly)

poripsa, 2021-11-05:
## Bug report

**Issue details**

Flashing the firmware on Pixhawk (CubeOrange) with different default parameters (rfnd, batt_monitor, ATC params set) results in some of these parameters not getting set up properly. This has been tried with QGC and MissionPlanner.

Previously the same defaults were set up for Copter-4.0.7 firmware and it used to flash correctly.
**Version**
Copter-4.1.0

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[defaults.txt](https://github.com/ArduPilot/ardupilot/files/7486613/defaults.txt)

[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
hex

**Hardware type**
Pixhawk CubeOrange

**Logs**

khancyr, 2021-11-05:
hello,

how do you set up those parameters ?

poripsa, 2021-11-05:
I edit the defaults.parm file here : https://github.com/ArduPilot/ardupilot/blob/Copter-4.1.0/libraries/AP_HAL_ChibiOS/hwdef/CubeOrange/defaults.parm

run: ./waf configure --board CubeOrange
and: ./waf copter 
(from root ardupilot)

then use QGC to select the arducopter.apj file in build/CubeOrange/bin to upload firmware (in its firmware section)

IamPete1, 2021-11-05:
Was the Cube reset to defaults? Either by flashing to plane first or with the param reset button.   AP defaults will not override parameters that have been previously configured, even if they are still the default value.

poripsa, 2021-11-12:
Yes it was. It was also flashed on a brand new pixhawk, had same issues.