ArduPilot/ardupilot #19175: [VTOL] Q_TAILSIT_GSCMSK it also acts on the PIDs of the Plane

robustini, 2021-11-06:
Tested with the current master in Realflight with CAT Tailsitter.
This scaling should in my opinion only work for the copter part, not performing any scaling on the plane, but I have evidence that something touches instead.
In the attached log I make changes to this parameter during a simple automatic mission.
Roll attitude of the plane without scaling:

[image]

With Q_TAILSIT_GSCMSK set to 4:

[image]

robustini, 2021-11-06:
The log: https://drive.google.com/file/d/1EqvC65qi4Ipb0kOeNwc3Cglyc1iD8eFY/view?usp=sharing

robustini, 2021-11-06:
Another small log, in the first part parameter set to 4 and then to 13, the problem is evident.
Then set to 15, the problem disappears.
Rightly this goes to alter the VTOL, but it is evident that something also happens in the Plane. 

[00000003.ZIP](https://github.com/ArduPilot/ardupilot/files/7490487/00000003.ZIP)

IamPete1, 2021-11-06:
Looks like your just in Q assist, it is expect behavior to use Q params when in assist. 

[image]

You can see that as soon as your airspeed increases above your set `Q_ASSIST_SPEED` of 11 assistance stops and the oscillation goes away. 

So either setup the Q params so there is no oscillation, or reduce/turn off `Q_ASSIST_SPEED`

closed 2021-11-06 by IamPete1