ArduPilot/ardupilot #19190: Write autotest for the no-short-failsafe option in auto mode for Plane

peterbarker, 2021-11-09:
## Feature request

**Is your feature request related to a problem? Please describe.**
We had a regression where a user set the ignore-short-failsafe-in-auto option and it didn't have an effect

**Describe the solution you'd like**
An autotest to ensure the option works

https://discuss.ardupilot.org/t/fs-long-actn-and-fs-short-actn/77900/8

**Describe alternatives you've considered**
Having the feature break again and we don't know about it

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

khanasif786, 2022-01-05:
@peterbarker sir, In my case quadplane doesn't go in circle during FS_SHORT_TIMEOUT time, No matter what mode it is currently in. I set FS_SHORT_ACTN  to  1.

Bhawna1203, 2023-01-24:
Hii @peterbarker Can you please explain the issue in more detail?
So that I can work on this

peterbarker, 2023-02-22:
@ConnectBhawna I answered much the same question in Discord:

```
@Shashwat Ganesh
@Peter Barker In this issue https://github.com/ArduPilot/ardupilot/issues/19190, you mentioned that the test should ensure that the "ignore-short-failsafe-in-auto option" has an effect. But in the linked discussion (https://discuss.ardupilot.org/t/fs-long-actn-and-fs-short-actn/77900/8), the user reported that setting FS_SHORT_ACTN to CIRCLE didn't produce an effect. I am fine with including tests for both, just in case. Is that fine? Apologies for re-raising a nearly 2 year old issue. (edited)
[9:16 AM]Peter Barker: Well - we did fix the bug.  So unless we've managed to re-introduce the bug since, that option should work.  The test is to ensure we don't break that option.  Again.
[9:16 AM]Peter Barker: ... this is kind of what regression tests are for 
```

andyp1per, 2025-10-26:
> Hi, [@peterbarker](https://github.com/peterbarker) could you please assign this issue to me? I’ve reviewed the project’s Code of Conduct and Contribution Guidelines, and I have the required skills and understanding of the tech stack to work on this effectively. I’ve already started analyzing the codebase and begun initial work toward resolving the issue. I’m confident I can implement a clean, optimized, and maintainable solution that aligns with the project’s standards. Looking forward to contributing!

I suggest you ask for a single issue to be assigned to you and then resolve it without it being obviously LLM generated slop. At the moment you give the impression of just firing stuff into ChatGPT and copy and pasting the answer