ArduPilot/ardupilot #19238: Rover/Boat: support bow thruster

AlksSAV, 2021-11-13:
## Feature request
Manoeuvering thruster (bow thruster or stern thruster) is a transversal propulsion device built into  to the bow or stern, of a ship or boat to make it more manoeuvrable.

This innovation will add even more features for the LOITER mode.
I have a Pixhawk-based ship model, the thruster only works in manual mode

It seems to me that the execution is similar to: 
    - “Vectored thrust” (only the deviation angle has two fixed values - or + 180 degrees) 
    -  or ROV with a lateral thruster.
 
**Describe alternatives you've considered**
Previously I created a [topic ](https://discuss.ardupilot.org/t/add-support-bow-thruster-for-boat/77780
)in discuss ardupilot
# 
**Platform**
[X] Rover

**Additional**
[image]

geofrancis, 2021-11-13:
A bow thruster is already supported as a yaw device, and to transverse it can be used as a motor in an omni 3 configuration. I don't see how having it has a separate thing is going to add anything? Like what scenarios should it be operational that is not covered by a yaw device or the omni 3?

AlksSAV, 2021-11-13:
> A bow thruster is already supported as a yaw device, and to transverse it can be used as a motor in an omni 3 configuration. I don't see how having it has a separate thing is going to add anything? Like what scenarios should it be operational that is not covered by a yaw device or the omni 3?

Maybe I expressed myself incorrectly earlier. I want to make the bow thruster work like on real ships. That is, the bow thruster does not work constantly, but only then it is turned on by the operator. But at the same time, I would like the autopilot to be able to use this device during the mission, and other drive modes(loiter, auto etc).

I will try on your recommendation, there is another thruster. I am confused by the fact that the thrust vector of the stern motors is not the same as on my model.

geofrancis, 2021-11-14:
> Maybe I expressed myself incorrectly earlier. I want to make the bow thruster work like on real ships. That is, the bow thruster does not work constantly, but only then it is turned on by the [...]

So my understanding is you essentially want to selectively enable the bow thruster at low throttle to help with yaw. I think someone done a script very similar to what you're wanting to do, but they were just adding thrust to the main motor to help the boat to turn, you would just need to change that to the bow thruster, I believe they had it setup so it was only active if the main  throttle was below a specific level so essentially when the rudder had the least authority.
https://discuss.ardupilot.org/t/improve-steering-if-motor-throttle-is-off/74572

I believe OMNI3 should work with your boat because your bow thruster is technically pushing the boat sideways but because it's offset from the centre of mass it will cause it to yaw, but that can be counteracted by the main propellers running on opposite directions, leaving a net force pushing you to the side. It will also be assisted by your rudders ability to vector the thrust from the rear of the boat.

rmackay9, 2021-11-14:
I'm sure we had a request for this before in the issues list but I can't immediately find it.  This seems like a valid enhancement request to me.  Auto, Guided, etc don't yet make use of a boat's ability to move laterally even on the Omni vehicles.