ArduPilot/ardupilot #19260: Multiple serial protocol 23 not allowed

Hwurzburg, 2021-11-15:
While setting up a MatekF405-WSE to replace a defective 765-Wing, I discovered that having two serial ports set to protocol 23 does not work....
I set CRSF on Telem1 (Serial1) of the F405-wse and it was not being detected....I had transferred params from the previous F765-Wing installation which had CRSF on UART6 (alt config 1) protocol 23 Serial7.....and had removed the alt config (although it should not be required since I was not using UART2 for which the alt config would enable on the F405-Wing)....but did not change SERIAL7 protocol from 23....this would have UART2 fully active and not matter since its not being used

CRSF on SERAIL1 cannot be detected until  I removed protocol 23 from SERIAL 7...also tried setting SERIAL4 to 23, and it did not allow detection on SERIAL 1 of the real CRSF connection....I believe this is a bug...we should be able to have two serial RX in the system...if not we should warn in the wiki
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

andyp1per, 2021-11-15:
No, this isn't a bug. You can only have multiple RX's if one is on the IOMCU - you cannot have multiple on UARTs, the RCProtocol singleton prevents this. We should change the wiki.

Note this is true of a bunch of things, e.g. ESC telemetry

peterbarker, 2021-11-22:
On Mon, 15 Nov 2021, Henry Wurzburg wrote:

> CRSF on SERAIL1 cannot be detected until I removed protocol 23 from SERIAL 7...also tried setting SERIAL4 to 23, and it did not allow detection on SERIAL 1 of the real CRSF connection....I believe [...]

I know this was closed as not-a-bug, but I think it could reasonably stay 
open as a feature request.

Hwurzburg, 2021-11-22:
done