ArduPilot/ardupilot #19285: Mission speed and HDOP issues

TexoDSI, 2021-11-17:
## Bug report

**Issue details**

Speed of aircraft is not what has been commanded in mission planner, we set speed at 2.5 and 5m/s and it was traveling between 8m/s and 10.5m/s.  As soon as the mission is started it takes off at speed.  

We are also having HDOP issues at 2 different locations on 2 seperate days, it is constantly over 1.2 and gets close to 2.  We did not have issues with this prior to updating to the most recent version of firmware and mission planner.

**Version**
4.1.1

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X8

**Hardware type**
Orange Cube

**Logs**
[2021-11-16 11-52-56.log](https://github.com/ArduPilot/ardupilot/files/7555487/2021-11-16.11-52-56.log)
[2021-11-16 12-26-24.log](https://github.com/ArduPilot/ardupilot/files/7555488/2021-11-16.12-26-24.log)
[2021-11-17 11-56-55.log](https://github.com/ArduPilot/ardupilot/files/7555489/2021-11-17.11-56-55.log)
[2021-11-17 12-50-20.log](https://github.com/ArduPilot/ardupilot/files/7555490/2021-11-17.12-50-20.log)

amilcarlucas, 2021-11-17:
HDOP issues are not caused by firmware update. HDOP is a measure of the GNSS satellite constellation geometric positions.

So unless the firmware caused the GNSS satellites to shift their orbits... it is not a ArduPilot bug.
The real solution is to improve the GNSS signal quality, by moving the antenna away from electronic interference.

The UAV speed is probably an issue though.

rmackay9, 2021-11-18:
Normally it is best to raise support issues in the support forums so I will close this issue.  Sorry to be strict but if we're not then our to-do list (which is what this issues list is for) becomes another support forum which will make it difficult to track known bugs and enhancements requests.

Here is the link to the support forum for Copter-4.1. https://discuss.ardupilot.org/c/arducopter/copter-41/159

The WPNAV_SPEED parameter is set to 1000 (10m/s) so this is what it will use by default.  I see the DO_SET_SPEED is right at the beginning of the mission (and only appears once).  I suspect what has happened is the vehicle was switched into Auto mode earlier in the flight, perhaps when the vehicle was disarmed.  Because MIS_RESTART = 0 when the vehicle is later switched to Auto mode it does not restart the mission from the first command but instead starts from where it left off which is the first WAYPOINT command (which it didn't complete earlier).  If you try again with MIS_RESTART = 1 I suspect it will act as you expected.

By the way, the vehicle has high vibration levels.  They don't seem to be causing a problem but they are in the grey zone (between 30m/s/s and 60m/s/s) which can lead to altitude and position control issues.
[image]

closed 2021-11-18 by rmackay9