ArduPilot/ardupilot #19318: Cube Orange cannot connect to MP/QGC etc when using Windows on a Mac with Parallels.

Angus-TStop, 2021-11-20:
## Bug report

**Issue details**

Cannot connect to Cube Orange (have tried various Cube Oranges) via USB on Windows 10 or 11 on a MacBook using Parallels. The flight mode indicator changes from a red 'unknown' to a white one but no connection occurs with the report being 'No heartbeat packets received'. The Cubes are correctly appearing in the Device Manager ie 'Cube Orange Mavlink (COM3)'. On the same computer with the same Cube and cables the connection works perfectly if Bootcamp is used. I have completely uninstalled Windows and Parallels and then reinstalled them but the problem persists. 

**Version**
All versions

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
All

**Hardware type**
Cube Orange

**Logs**
Logs not applicable to this issue

IamPete1, 2021-11-20:
This does not sound like anything we can fix in ArduPilot. A little googling suggests there is some setup in parallels that needs to be done. 

https://kb.parallels.com/6318

*I don't have a mac and have never used parallels.

Angus-TStop, 2021-11-21:
Thank you. I have googled that and done that but still no luck. I have asked on Cubepilot forums and Parallels but no luck there so far. Fingers crossed someone might be able to help. Its hard when the problem is at a crossroads of all involved systems :)

rmackay9, 2021-11-22:
This is really a support issue so I think I'll close this.  Sorry to be strict but otherwise our to-do list of bug fixes and enhancements turns into another support forum and we lose even more track of what code changes need doing.

closed 2021-11-22 by rmackay9