ArduPilot/ardupilot #19320: Remove parachute deployment limit.

FlyGuyBry, 2021-11-20:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REQUEST WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://disc
## Feature request

Please remove the limit for parachute deployments.  There is no reason to limit this.  If it does deploy actuating again causes no harm. 

There is a bug that I'm told has already been fixed where parachutes deploy on boot.  This combined with the one deployment limit broke parachute all together.  

Regardless of that bug, it is very useful to check key safety features like parachutes during preflight.  Between that and the boot 3 deployments per flight would be normal for me.  One boot, one test holding door, and one actual deployment.

**Platform**
[  ] All
[  ] AntennaTracker
[X  ] Copter
[X  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

khancyr, 2021-11-21:
Hello,

Which limit are you speaking about ?
Altitude ? Angle ? Freefall detection ?
The altitude limit could be removed by parameters

But using the parachute isn't harmless depending on your operation context.
The parachute just reduce the ground impact force but don't prevent the free fall. And under some altitude it is useless. Triggering the parachute could also lead to have it going into power line or some other dangerous place, or move the vehicle chute.

About you operation context, I assume that you have a servo based parachute, unfortunately we need to take into account pyro based parachute that cannot be check on preflight.

FlyGuyBry, 2021-11-21:
I'm referring to the one deployment per boot limit.

peterbarker, 2021-11-22:
On Sun, 21 Nov 2021, FlyGuyBry wrote:

> I'm referring to the one deployment per boot limit.

Are your CHUTE_OPTIONS 0?