ArduPilot/ardupilot #19327: Synthetic Airspeed being miscalculated in QuadPlane with Airspeed disable RC switch

Hwurzburg, 2021-11-22:
Using RC OPTION for disabling airspeed use....started flying in VTOL, with AIRSPEED_USE disabled by RC switch..my first attempt to FW transition failed, and even after I switched the switch off, and changed modes to CRUISE,etc... it remaining is AIRSPEED_WAIT....this is due to Synth AS being 0 or near 0 ,even though vehicle is at flying speed, and GPS velocity is good...had to disable switch then move back to VTOL and then Synth AS started working and was able to do a normal transition to fixed wing

will try to duplicate in SITL

[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

Log: https://www.dropbox.com/s/xd0cyc4iaygsgr5/airspeed%20rc%20switch%20transition%20failure.BIN?dl=0

Hwurzburg, 2021-12-07:
here is another log using the ARSPD_USE param itself and causing a 0 synth airspeed
https://www.dropbox.com/s/mtf5q93df7h41jt/SAs%20failure%232.bin?dl=0

still trying to get it in SITL.... this was with master ~ 11/08/20 commit 6c29eb73d, as above

peterbarker, 2021-12-08:
I've created an autotest here which doesn't fail:  https://github.com/ArduPilot/ardupilot/pull/19461

Any hints?

peterbarker, 2021-12-08:
@Hwurzburg to get a good wind estimate you usually have to turn some corners - maybe that's the problem?

Hwurzburg, 2021-12-08:
I do not think in general it should...there is high ground speed and I also stopped and turned around in VTOL and tried to transition in the opposite direction....its true that trying to transition into a high wind can stop the transition due to LOW groundspeed, hence my PR #19326....but this happened with no wind in the first log....and CTUN.SAs is non zero even in those cases...just low
a few conditions not in your new autotest:
1. airspeed use is disabled from the start of init thru arm and takeoff and transition attempt
2. yaw is not aligned in EKF3
3. MAG_EN is 0
my WAG guess is that @priseborough mods for compass-less, airspeed-less wind estimates for EKF3 are somehow involved...perhaps making SAs get latched into a negative value and constrained to zero even at high ground speeds

peterbarker, 2021-12-09:
My current best-guess is some sort of screwup with the fly-foward assumption.

Note that `CTUN.SAs` always comes from DCM.  Unless you have a compass, or a GPS and fly-forward then you can't get a synthetic airspeed.  This is a Tri - I wonder if it is playing games with fly-forward and we're not actually flying forward.  `Plane::update_fly_forward` looks complicated enough there might be a hole in there.

Hwurzburg, 2021-12-15:
@tridge was able to get native SITL to do it on its quadplane  https://www.dropbox.com/s/sc7pjjavkcs54mf/SAa%20fail.BIN?dl=0

priseborough, 2021-12-23:
Can you please reproduce with LOG_DISARMED=1 and LOG_REPLAY=1 so i can investigate and fix the root cause using the EKF replay. TXS.

peterbarker, 2022-08-16:
@Hwurzburg nudge on this one

Hwurzburg, 2022-11-27:
havent duplicated...closing

closed 2022-11-27 by Hwurzburg

DaveProMax, 2026-11-14:
I'm experiencing almost exactly the same issue. During the transition, the synthetic airspeed drops to zero. It seems that this bug may still be present.
If you have some time, would you be willing to take a look at my flight logs and help investigate what might be causing this? I would really appreciate your time and help. Thank you very much!

Hwurzburg, 2026-11-14:
> I'm experiencing almost exactly the same issue. During the transition, the synthetic airspeed drops to zero. It seems that this bug may still be present. If you have some time, would you be willing [...]

open a new issue with an attached dataflash log showing it