ArduPilot/ardupilot #19346: Rover: DSHOT reverse range is inverted

geofrancis, 2021-11-23:
When trying to use Reversable Shot motors with rover firmware the reverse values are inverted.

0-100% forward should be 1500-2000us output and that works fine but 0-100% reverse ends up as 1000-1500us so as minimum speed reverse is 1000us and full speed reverse is 1500us rather than minimum reverse at 1500us and full reverse at 1000us 

Regular PWM mode works as expected but it disables ESC telemetry.

rmackay9, 2021-11-23:
FYI @andyp1per

geofrancis, 2021-11-24:
is there any specific reason Dshot needs to be enabled to get blheli32 telemetry?

andyp1per, 2021-11-24:
> is there any specific reason Dshot needs to be enabled to get blheli32 telemetry? or is it just assumed that if they have a telemetry capable esc they want to use dshot?

Yes, requesting telemetry is part of the dshot protocol - it is not part of any of the other protocols. It's possible, I suppose, that BLHeli's auto telemetry feature might help here, but we are not currently geared up to cope with that.

andyp1per, 2021-11-24:
I'm having a hard time believing this is different in 4.2 - the dshot code for 4.2 and 4.1 is very little changed and the place we convert for reversing motors is almost identical. Please can you post your full parameter list so that I can test?

andyp1per, 2021-11-24:
@shellixyz you are using reversing motors on 4.2 I believe?

geofrancis, 2021-11-24:
I'm trying to replicate it, but I can't even 4.2 to start any more, it's just stuck at gyro initializing. flashed back to 4.1 copter reset all settings to try and clear the board but when i flash 4.2 rover i just get repeated message of "initializing". 4.1.1 rover is working fine . but i just cant get 4.2 to boot anymore. il get you a parameter list as soon as i can get it started again.

rmackay9, 2021-11-24:
@geofrancis,

The "Initialising ArduPilot" message should be fixed if you rebase on master.  There was a bad PR that was merged yesterday but the issue has been fixed now.  Sorry for the troubles.

geofrancis, 2021-11-24:
flashing 4.2dev via mission planner is still giving me the error.

rmackay9, 2021-11-24:
@geofrancis, yes, the build server hasn't gotten through rebuilding all the various versions yet.  It should be fixed in a few hours though.

shellixyz, 2021-11-24:
> @shellixyz you are using reversing motors on 4.2 I believe?

I do yes, works fine for me with Plane and BLH32

geofrancis, 2021-11-24:
I think its something to do with the mixing in rover, as the pwm values showing on mission planner are showing the correct values and that  they are where supposed to be, but the dshot output doesn't match what its showing.

geofrancis, 2021-11-27:
I have just done a fresh installation of 4.2, configured everything from scratch back to the way it was working in 4.1 and the problem is back, reverse is inverte
[boat.zip](https://github.com/ArduPilot/ardupilot/files/7610986/boat.zip)
d.

andyp1per, 2021-11-27:
What flight controller is this on?

geofrancis, 2021-11-27:
pixhawk 1

geofrancis, 2021-11-29:
were you able to replicate this?

geofrancis, 2021-11-29:
> Sorry, was doing other stuff over the weekend

no problem

rmackay9, 2021-12-14:
We discussed this at the dev call today and I agreed I would try and reproduce this on a BLHeliESC that I have.

AlksSAV, 2022-01-31:
I have the[ same problem](https://discuss.ardupilot.org/t/blheli-32-missing-parameter-servo-blh-bdmask/81064/21?u=mckey) with DShot on Rover 4.1.3 firmware. Previously there was version 4.2
My Equipment:
 - Pixhawk 2.4.8
 - BLHeli_32 DYS Aria 35A

Jahor, 2022-03-15:
Same issue. Also in BLHeli_32 when configured in Bidirectional 3D (not Soft) sending 1048 (idle state) over DShot causes motors to spin. May be it is possible to send command 0 (stop) when idle?
When sending 1500 with PWM it does not happen, as they have some dead-band.

rmackay9, 2022-03-22:
I've reproduced the issue on a CubeOrange running Rover-4.2.0-beta and 4.3.0-dev (aka latest, aka master) so hopefully we can get to the bottom of this.

rmackay9, 2022-03-22:
Ok, I think we have a fix for this issue here https://github.com/ArduPilot/ardupilot/pull/20349.

@geofrancis do you think you could test this resolves the issue?  It does for me but I'd like to get it tested with various different autopilots if possible.

geofrancis, 2022-03-29:
I have been ill for the last month, I haven't been able to test anything yet unfortunately.

rmackay9, 2022-03-29:
@geofrancis, hope you're feeling better soon.  I've tested it quite thoroughly and the fix is included in -beta3 which should be out later today.

I think I might close this one but we can re-open if you find the problem still exists.

closed 2022-03-29 by rmackay9

AlksSAV, 2022-04-14:
Now I have tested on version 4.3 dev, still not correct work. Is it just me who has such a problem?

andyp1per, 2022-04-14:
> Now I have tested on version 4.3 dev, still not correct work. Is it just me who has such a problem?

Please post your params

AlksSAV, 2022-04-20:
> > Now I have tested on version 4.3 dev, still not correct work. Is it just me who has such a problem?
> 
> 
> 
> Please post your params

The problem turned out to be in ESC, tested everything works. 

Pixhawk 2.4.8, DYS Aria blheli 32.

rmackay9, 2022-04-20:
@AlksSAV,

Great! thanks very much for the feedback.

LimitlessGreen, 2024-10-30:
Have an Bluerobotics Basic ESC with Bluejay on Rover V4.5.7. Set the SERVO_BLH_3DMASK and have inverted reverse throttle. (Altough the servo output in Missionplanner looks normal)

rmackay9, 2024-10-31:
Hi @LimitlessGreen,

Could you post an onboard log file?  Ideally these issues should reported in [the forums ](https://discuss.ardupilot.org/c/ardurover/arduboat/54).  Have you reported there yet?

geofrancis, 2024-10-31:
> Hi @LimitlessGreen,
> 
> Could you post an onboard log file? Ideally these issues should reported in [the forums ](https://discuss.ardupilot.org/c/ardurover/arduboat/54). Have you reported there yet?

I tagged you about this in july 
https://discuss.ardupilot.org/t/dshot-reverse-inverted/120439/4

andyp1per, 2024-10-31:
Please try 4.6