ArduPilot/ardupilot #19361: Rover: INITIAL_MODE broken?

rmackay9, 2021-11-25:
During a quick SITL test while investigating [this user report](https://discuss.ardupilot.org/t/some-doubts-about-ardurover-ros2-and-wheels-encoders/78777/2) I found that Rover's INITIAL_MODE doesn't seem to work.

I set the parameter to 1 for Acro or 15 for Guided and it simply didn't  end up in these modes when the simulator was restarted.  I haven't investigated this thoroughly.

shiv-tyagi, 2021-12-07:
@rmackay9 
I was able to reproduce this. I would love working on this issue. Will push a patch if able to get to the bottom of this problem.

khancyr, 2021-12-08:
it would be great to improve the autotest on this as it should definitively work. Most probable that the mode change is refuse because of missing position or velocity estimate

shiv-tyagi, 2021-12-11:
@rmackay9 I investigated this issue. I found out that the rc switch (6 pos) we have to set flight mode is overriding this setting. Initially the rover successfully set itself to the mode specified in `INITIAL_MODE` but as soon as it starts getting rc pulses, it reads the mode switch and sets it to the mode which rc is pointing to.

rmackay9, 2021-12-14:
@shiv-tyagi,

Great, thanks very much for this.  I think I'll close this issue then.

closed 2021-12-14 by rmackay9